jeelabs/esp-link #210: Empty wifi configuration after a while.

hetii, 2016-10-30:
I notice some odd issue regarding wifi configuration.
It looks like after a while or after some unknow condition esp-link lost his wifi settings.
In terminal esp is set to AP mode and with all IPs set to 0.0.0.0
Even when I reflash it using user1.bin and blank.bin the settings is still empty and don`t switch back to AP with DHCP to be able to connect to it on 192.168.4.1

Only by calling `esptool.py erase_flash` and reflashing, configuration is set to default one.

minida28, 2016-10-31:
I think I remember I had this situation before (or similar; can't remember exactly).
Do you mind try this:
1. Flash the esp using older esplink version - I used version 2.1.7
2. Then update to the latest firmware using wiflash
Steps above solved my problems (unable to set IP address manually or via DHCP).

tve, 2016-11-30:
Mhh, I'm not sure how to reproduce this, any ideas?

tve, 2017-06-18:
closing 'cause nothing has moved here in a very long time.

closed 2017-06-18 by tve