jeelabs/esp-link #211: sending message to Telegram

karabas, 2016-11-07:
My Esplink is connected to office router and via serial to Arduino. I need to force my arduino send a message or signal to esplink and esplink after receiving must send a preconfigured message to Telegram.
How can I do this?

SloMusti, 2016-11-20:
This works for me: https://github.com/Gianbacchio/ESP8266-TelegramBot

Note that this is not using esp-link but treating esp8266 itself as an Arduino device. You may either get rid of your Arduino board and connect devices directly to ESP or get a serial communication going.

tve, 2016-11-30:
I'm not familiar with telegram. If it's an HTTP API you can use the REST support in esp-link. I would recommend that you describe what you're trying to do in the esp-link gitter chat. https://gitter.im/jeelabs/esp-link

closed 2016-11-30 by tve