jeelabs/esp-link #214: flow control ro any other means to tell esp is connected(from the AVR side)?

solidboredom, 2016-11-11:
First i want to join the chorus and to thank you for the the esplink.
it is developing fast, and the issues are vanishing very fast as well, and you have been very supportive.
really a ton of "thank you" for that.
now the issue:

if was experiencing blocking in my code and was assuming it is because my Arduinos Serial.send () had blocked the code when the WIFI connection was bad or lost.

so i have started to look at the state of the connection by Serial.availableForWrite(), looking at how many bytes are free in the buffer and switching my AVR code to the "OFF-LIne" mode.

today i have read through a somewhat related issue  at the project and it seems to me,
 the connection to AVR has no flow control at all (i assumed a XON/XOFF or alike). that explains why the Arduinos serial write buffer changes unpredictably independet of either ESP is OFF or ON-line with WIFI router.

so i am now trying to figure a way to switch my AVR code to the OFF-Line-state once the connection is lost and back flushing the collected data once it is restored.
the blocking of the code has probably come from other source, and the AVRs serial.send() has probably nothing to do with it, right?

1. are there any circumstances under which Arduinos serial.send() connected to ESP might block (serial buffer full)?

2. is there any other way i can tell the connection is still there? (i have no more access to the hardware it is in a water- tight confinement, sigh)

3. i just have come up with the light-house-ping idea: to use a script on the computer to ping the AVR with a ping-text-message. judging on how much time have passed since the last ping arrived, AVR could judge if it is still connected or not. 
Any better/simpler solution ideas without access to the Hardware?
thanks in advance for your precius help.

tve, 2016-12-11:
(Sorry, I overlooked your issue)
You are correct that there is no flow control. TX from the arduino will never be blocked. The esp-link should respond to the commands whether it is connected to wifi or not, however. I think your ping idea is the best way to go, because it gives you an end-to-end check. You can initiate the ping on either end, e.g., you could have something send your AVR a message periodically or you could have your AVR send a ping periodically and expect a response back.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve