jeelabs/esp-link #215: Virtual TTY for esp-link with ioctl support.

hetii, 2016-11-13:
Hello ;)

I wrote application that expose virtual tty under linux (/dev/ttyCUSE0) and pass all data to esp-link.
It also handle IOCTL so its possible to control esp-link uart settings like in any other serial port.

The code can be found on my bitbucket account:
[cusetty](https://bitbucket.org/hetii/cusetty/)

Its in alpha stage and currently handle uart communication as well as `fmt` rest argument.

But it should be easy to add as well other things like setting baudrate RTS/CTS flow etc...

Feel free to contribute it :)

tve, 2016-11-30:
Cool! Do you use the telnet escape codes supported by esp-link for the uart settings?

tve, 2016-12-11:
Nice! I think it would be better if you used the Telnet protocol escape codes to change settings in esp-link, though.

hetii, 2017-05-24:
I didn`t know before  that esp-link support escape sequences that change uart settings.
Now I see some implementation in telnetUnwrap() function.
Is it based on RFC2217 or some other?

tve, 2017-06-18:
yes it is.