jeelabs/esp-link #218: Scanning for networks indefinitely

unlimitedbacon, 2016-11-21:
On a freshly flashed module, when you go to the WiFi station page to connect to a network for the first time, it shows "Scanning..." indefinitely. Refreshing the page doesn't help. The debug log doesn't show a network scan happening.

The scan doesn't actually happen until you switch to STA+AP mode (the default is AP). However, if you switch back to AP again after that, scanning still works as expected.

[Here's a log](http://pastebin.com/KWCVdbQa)

tve, 2016-11-21:
thanks for flagging this, I think I've seen this too.

WheresWally, 2016-11-27:
Yes I get this too. On a new flash of a ESP8266-01, the debug output shows the same 3 lines, sorry I didn't grab them. Default mode shows as STA+AP but i suspect it isnt. I have to go to WiFi Soft-AP Configuration page and switch to STA mode only, which means I loose connection as I haven't connected it to my home network yet and am connected to 192.168.4.1 
But then I power cycle it, and the then it will scan and find and can be associated as expected. 
I did the 3 times today after a reflash V3 alpha and then then stable version 2.3 (or what ever it is)

closed 2016-12-12 by tve

susanin-ivan, 2018-04-09:
Hi,

Thanks a lot for great and usefull software!

I've got similar issue, but slightly different - debug do show found networks, but still none on web page:

275432> HTTP GET /wifi/scan: 200, 7ms, h=10760
276511> GET scan: cgiData=0 noAps=34
276665> scandone
276666> Scan done: found 33 APs
276666> bss1: DIRECT-76-HP DeskJet 4670 series � (-79) <--- this one seems to be issue
276666> bss2: Ora (-85)
276668> bss3: HOTBOX-3A76 (-94)
276671> bss4: HOTBOX-709A (-90)

Firefox's log:

JSON parse error: SyntaxError: JSON.parse: bad control character in string literal at line 2 column 45 of the JSON data. In: {"result": {"inProgress": "0", "APs": [ {"essid": "DIRECT-76-HP DeskJet 4670 series��", "rssi": -80, "enc": "3"},

Again, thank you!

commit 5efbf326 (2016-12-12): fix scanning after clean flash, fixes #218