jeelabs/esp-link #219: Upload the HEX file to the ESP_LINK and then flash the attached uController please?

WheresWally, 2016-11-27:
As a possible new feature, seeing as we now can flash the ESP firmware, wouldn't it be possible to use the same mechanisms to upload a HEX file as is done by the flash firmware and by the webserver, and then use the same routines for flashing the attached ucontroller?
I hope this can be a feature as it is more intuitive than using avrdude or curl.
Terry

tve, 2016-11-30:
Yes, this would definitely be possible. I'm not sure it's that convenient. Every time you make a change and need to re-upload you end up having to click around a bunch to perform the upload in your browser. But I'm certainly open to a PR :-)

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve