jeelabs/esp-link #220: RESET Pin high impedance when quiescent please

WheresWally, 2016-11-27:
Possible bug fix or improvement.
I have a MEGA with a watchdog chip as well as the ESP_LINK. The watchdog chip cant reset the MEGA , but the ESP_LINK can. The watch chip wire is in a high impedance state when not needed.
If I remove the Reset wire from the ESP_LINK (on a ESP8266-01) then the watchdog does work.
I installed a diode between the ESP-LINK Reset and the MEGA, then the wacthdog chip and the ESP_LINK can both reset the MEGA as desired.
Like this .
MEGA.Reset  ---|>| ---- ESP_LINK.Reset.

Does the ESP_LINK.Reset pin go to a high impedance (an input) state when not needed to perform a reset?
If not, can it be made to do that please?
Terry

tve, 2016-11-30:
This could definitely be implemented. A PR would be awesome ;-)

WheresWally, 2016-12-02:
I am not sure what a Pull request requires of me.
I have so far found that in the file 'serbridge.c,' the lines with  **GPIO_OUTPUT_SET(mcu_reset_pin, 1);** will most likely need to be changed to set the pin as an input. Not sure how to do that yet.

tve, 2016-12-03:
Yes, that should work, but you also need to re-enable the pin when it's set to 0. No need for curly braces.

A PR is more involved 'cause this would need a setting in the UI and be persisted in flash. You can of course just make the changes locally for yourself ;-).

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve