jeelabs/esp-link #221: SLIP overhead

C-Artur, 2016-11-28:
Dear Contributors,

The esp-link is a great piece of software, I very like all its features, but have one tiny issue with communication from Arduino Nano v3 (atmega 168/1KB SRAM) - the el-client overhead (SLIP etc.).
It takes about 30% of atmega168 RAM, when I try to use MQTT communication.

Can you please provide any info how to communicate with esp-link directly using Serial library?

tve, 2016-11-30:
You do not have to use the SLIP encoding to use esp-link as a simple serial bridge. Just disable SLIP in the esp-link web interface. You only need to use SLIP to give esp-link commands from the arduino side. If you need to do that, you can reduce the buffer size in the SLIP library on the arduino side. I hope this helps.

tve, 2016-12-11:
I don't think there's anything for me to do here, let me know if you have further questions.

closed 2016-12-11 by tve