jeelabs/esp-link #223: ATO programming of Arduino MEGA, does anyone have this working over ESP-LINK V3?

WheresWally, 2016-11-29:
I have ESP8266-01 with V3 ESP-LINK
The baud is universally set to 115200.
I can flash a UNO 100% of the time using the the HTTP upload method.
Fails when using the HTTP method to a MEGA 

```When trying to upload to a MEGA I use
avrdude -Cavrdude.conf -D -V -patmega2560 -carduino -Pnet:192.168.0.20:2323 -b115200 -Uflash:w:BrydenWeatherMC-MEGA.ino.hex
```

and get ...

```
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x62
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x66
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x72
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x4d
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x6d
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6f
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude done.  Thank you.
```

9600, 2016-11-29:
Hello, you called?

WheresWally, 2016-11-30:
I did call, and I said "Help me Please"
Any idea how to make the ESP-LINK program an attached uController  Arduino MEGA

CharlyBrok, 2016-11-30:
Is it the same issue as https://github.com/jeelabs/esp-link/issues/87 ?

WheresWally, 2016-11-30:
Yes it is, Thanks for linking to issue #[87](https://github.com/jeelabs/esp-link/issues/87) ?

WheresWally, 2016-11-30:
oops

ramarro123, 2016-11-30:
@tve it's this project still maintained?

tve, 2016-11-30:
Yup, but it's "feature complete" for my needs, so I don't have a ton of time to add new features. The last round of new features have been added by others and I integrated them. I don't have any MEGA so I'm unlikely to be able to add MEGA programming myself..

lunix80, 2016-11-30:
What a pity! With the use of MEGA it will be super!

tve, 2016-11-30:
If you feel this way, it would be awesome if you could contribute a pull request ;-)

tve, 2016-11-30:
This is a duplicate of #87

closed 2016-11-30 by tve