jeelabs/esp-link #224: Serial console to a Raspberry Pi

schoerg, 2016-12-02:
I have loaded the esp-link onto an ESP8266, working, mostly.

I connected the UART to the UART of a Raspberry Pi.

It's working with the webinterface, I can send normal Linux commands, I have to uncheck the "CR" box, otherwise I get double lines on every Enter key.

If I connect via telnet however, I have all sort of problems:
Using putty, a CR and LF is sent for every Enterkey, resulting in double lines. The Tab key is also not working. Is this client related? If yes I won't bother you, I'd rather fix it on my end.

tve, 2016-12-03:
The CR-LF is a setting in putty on the "Terminal" setting page. esp-link does not add/remove any cr or lf when using telnet. The tab key not working is puzzling, but again, nothing in esp-link interprets tabs.

schoerg, 2016-12-04:
Thanks, the settings have unfortunatley no effect, two newlines will be sent with every Enter key. But I now know this is client related, thanks.

tve, 2016-12-11:
You need to turn local echo off explicitly. I added the following to the README, let me know whether that solves your issue:

> If you are using esp-link to connect to the console of a linux system, such as an rPi, you
will most likely see what you typed being echoed twice. If you are on a linux system use
telnet and issue a `mode char` command (in telnet, hit the escape char `^]` and type `mode
char` at the prompt). If you are using putty on Windows, open the connection settings and
in the terminal settings set both `local echo` and `local line editing` to `off`.

closed 2017-01-02 by schoerg