jeelabs/esp-link #225: Pin/LED assignments (Adafruit HUZZAH Breakout)

rjblake, 2016-12-03:
I managed to get v2.2.3 loaded on a HUZZAH ESP8266, but noticed the onboard LEDs/GPIO assignments seem to be mixed up (only my opinion). Seems more logical to use the 'Blue LED' (GPIO2) for WiFi and 'Red LED' (GPIO0) for Serial (onboard). That being the case, the following would be reversed. Perhaps this is just limited to the HUZZAH board, however

tve, 2016-12-03:
> That being the case, the following would be reversed.
Did you want to create a PR or what are you referring to?

Also, the web UI lets you change pin assignments, did that not work for you?

rjblake, 2016-12-03:
My bad - change only took effect once I forced data through serial line. Working fine now I swapped. For the rest its cosmetics (gpio2/TX1), where TX1 has not relevance on my device. Thanks for awesome product by the way - much appreciated

tve, 2016-12-05:
thanks for the update, glad it's working now :-)

closed 2016-12-05 by tve