jeelabs/esp-link #227: Allow larger REST responses than 100 bytes

MaxForward, 2016-12-04:
I would like to use esp-link's REST functionality to get weather data from the internet.
However, the responses are quite extensive and sometimes contain far more than the first 100 bytes that esp-link supports. Just changing this line
`cmdResponseBody(pdata+pi, body_len>100?100:body_len);`
and increasing the 100 just helped a little.
I guess the other part of the problem lies here:
`// Receive HTTP response - this hacky function assumes that the full response is received in
// one go. Sigh...`

Thank you for your good work. I am using esp-link to flash my STM32 after having it recompiled for even parity on the UART. Now, a support for more extended REST responses would make it perfect to me.

tve, 2016-12-05:
Yup, as you saw it's a known issue. It's not simple to fix however and in the end, it's not clear this is the best avenue either. Just shipping tons of data to the uC and then having to parse it there is pretty resource intensive. You're using an stm32, so you probably have a bit more flash & memory than an AVR, but on an AVR there just isn't much space. Might be better to get some regexp or similar parsing into esp-link so it can parse out the piece of interest and forward that.

tve, 2017-06-18:
This has been implemented and should be supported in v3.2

closed 2017-06-18 by tve