jeelabs/esp-link #228: REST Requests from Python / Raspberry Pi

jn6118, 2016-12-08:
Hi,

Just to say that I think the esp-link firmware is brilliant and I think I can integrate this into many projects I am working on.  One of these, to which this question relates, is a raspberry pi zero based datalogger (scripted with python) to which I have interfaced a esp8266 - this will be installed at a remote location that does have access to a WiFi AP but I cannot know the SSID and p/w in advance.  As such, the esp-link firmware works really well to allow an on-the-fly setup.

I have flashed the esp-link firmware to the esp8266, have tested the practical aspects if using it. I can send strings over serial to the esp8266 using a python implementation of the SLIP protocol - see [.](http://recolog.blogspot.co.uk/2012/10/serial-line-internet-protocol-slip.html). These strings successfully appear in the serial log on the esp-link frontend.

I have looked through the esp-link library and program files relating to initiating REST requests but unfortunately my experience with c is fairly limited. It appears that a particular type of request is initiated over SLIP with an integer command, followed by a sequence of parameters as required. I would really like to implement this in python and for other reasons I cannot migrate this particular project to arduino.

What I was hoping might be possible was that a REST request could be initiated (and setup) with a series of text commands to the serial port, but I think it is probably lower level than this though? My needs for now are actually very simple - I am just trying to upload data to thingspeak and hence have a request of the form: **GET https://api.thingspeak.com/update?api_key=XXXXXXXXXXXXXXXX&field1=XXX** - i.e. entering the above into a browser without the GET would achieve the intended purpose. For now, I am not too worried about receiving feedback on whether the request was successful or not if it makes things easier.

Could anyone please point me in the right direction? Even a hacked together implementation in the first instance that works to prove the concept would allow me to settle on this firmware for the esp8266 and I could refine the script over time as I invest more time into it.

I am sure many people would benefit from a similar approach, if it is feasible. I really appreciate any help.

Thank you,
Josh

tve, 2016-12-08:
Thanks for your kind words. The message format isn't documented, oops. The request header is https://github.com/jeelabs/esp-link/blob/master/cmd/cmd.h#L23-L28 and each argument is https://github.com/jeelabs/esp-link/blob/master/cmd/cmd.h#L18-L21. I.e., a 16-bit command, a 16-bit number of args, a 32-bit bit token for the response, followed by the args. Each arg consists of a 16-bit length followed by the data bytes. I now forget whether the values are big-endian or little endian... At the end there is a CITT 16-bit CRC sum for the entire SLIP packet https://github.com/jeelabs/esp-link/blob/master/serial/crc16.c

You can follow the rest setup command parsing here https://github.com/jeelabs/esp-link/blob/master/rest/rest.c#L201-L233, it expects a hostname arg, a port arg (16 bits), and a "security mode" 1-byte arg which you can ignore (set to 0). Each REST request is parsed here: https://github.com/jeelabs/esp-link/blob/master/rest/rest.c#L343-L378

The responses come back with the 32-bit response token you pass in. This is used on arduino as the address of a function to call with the response buffer.

I know this is just a sketch description, sorry for that, I hope it gets you started.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve