jeelabs/esp-link #229: Bridge wifi to serial

silvius76, 2016-12-13:
Hi,

I would like to use esp8266mod ai-thinker with esp-link in bridge mode ( full trasparent serial ) for that:

[image]

I tested it in different configuration, but it didnt work
where I should connet (esp8266) serial pin like CTS, RTS, DTR, DSR ?

Regards

tve, 2016-12-13:
esp-link does not currently support flow-control, sorry.

silvius76, 2016-12-16:
Thanks

I have 2 possibility, software or hardware handshakle

SOFTWARE:
[image]

HARDWARE:
[image]

is possible with software flowcontrol ?

Regards

tve, 2016-12-18:
if end-to-end flow control through esp-link works for you, then it may be OK, but esp-link itself does not implement flow-control.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve