jeelabs/esp-link #230: Reset Arduino after finish the boot

FernandoGarcia, 2016-12-13:
Hi!

I can see that during the boot GPIO12 goes to down during a while but it is not enough to reset the Arduino.

I can see too that the "reset uC" function  in console.html works fine.

I would like to suggest you to call this function after finish the boot, so Arduino will reset together with ESP.

The best solution could be resync Arduino and ESP after a reset but the reset callback is not working as I told here: https://github.com/jeelabs/el-client/issues/29

Best regards.

tve, 2016-12-13:
esp-link does not make any attempt to reset the attached uC when it starts up. I don't know which function you set for gpio12, but if you put a pull-up on it it should stay up. Some users use the attached uC as master and esp-link as "slave" and power down the esp8266 to save power. In that use-case it would be very undesirable to reset the attached uC at esp-link boot time.
What are you trying to solve by resetting the arduino when esp-link boots?

FernandoGarcia, 2016-12-13:
I'm using default firmware available here.

I'm talking about function called by this button:

[image]

Maybe this one:

https://github.com/jeelabs/esp-link/blob/master/html/console.html#L81

It could be an optional function with check box in home page to enable.

The problem is if the ESP resets by any reason the Arduino won't resync and will keep trying send data. Arduino can crash for it. Call Sync() before try send data won't solve because there a delay to get the connection status.

tve, 2016-12-13:
If esp-link resets then the EL-Client library will resync. If not, please create a ticket on the el-client repo and describe the exact steps to reproduce the problem with debug output.

You're welcome to create a PR to add an option to reset the attached uC when esp-link starts up, but this is not something I plan to work on or something that can be on by default.

FernandoGarcia, 2016-12-14:
I would like to help you with a PR but I have no idea about what I have to edit on code.

Regarding el-client I have explained the problem for you here: https://github.com/jeelabs/el-client/issues/29

Thanks for your time.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve