jeelabs/esp-link #231: Telnet & HEX responses

rjblake, 2016-12-14:
Question (I guess), as can't seem to find more info elsewhere. I have the ESP connected to a Remeha boiler and in order to receive data, I need to send a HEX string to the ESP. I can do this from the *nix using nc, or through Telnet, etc. but the 'Microcontroller Console log'  gets filled up with all sorts of weird chars (well, expected). Is there anyway that these could either be translated back to a HEX code instead (e.g. 0xAB or \xAB), or even have an option to turn off the logging

tve, 2016-12-15:
There's no way to turn off the console log web page, but you can avoid going there ;-). It uses up 1KB-2KB of memory (I forget the exact size), other than that there's no harm done by having all these control characters "in there". So just ignore the page and continue using `nc` or your tool of choice to connect to port 23... Maybe I'm misunderstanding your question.

rjblake, 2016-12-15:
Ok. Not too concerned by all the control chars in the log. The other part of the question relates to the representation of the control chars - would it be a big effort to 'translate' these to a more human readable format? e.g. Instead of showing "▓æ╧"  in the log , an option to display 0xB2 0x91 0xCF instead? Would help debugging controller messages without the need to use additional tools (in my case)

tve, 2016-12-16:
It wouldn't be too hard to transform the text. There are already a couple of characters that are explicitly transformed, so it's "just" a matter of enhancing that: https://github.com/jeelabs/esp-link/blob/master/html/console.js#L34-L38
Pull request welcome ;-)

rjblake, 2016-12-16:
Sounds like a plan. I'll look into creating a PR and perhaps try to see what I can do over the coming break (although I'll be on holiday for a bit). Might look into adding a drop down selector on the page for HEX->ASCII, etc. Not sure if I can use the Javascript on uploaded 'Custom web pages' as make developing/testing quicker; but will give it a try and see where I can get to.

yesyesuk, 2016-12-16:
That would be fantastic. It would make troubleshooting easier for me too as I send binary data to / from my home automation devices using esp-link purely as a serial-WiFi bridge.

tve, 2016-12-18:
Using wiflash to update esp-link over wifi is actually pretty fast, about 15 seconds. If you want faster, you're best off experimenting using a local web page on your laptop with some static data.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve