jeelabs/esp-link #232: Tutorial

FernandoGarcia, 2016-12-16:
Hi @tve !

I did a tutorial to show how to set up a MQTT client to work with my controller but I think it can be useful to others users.

I spent so much time to understand how to make it work by myself so I hope it will make the life more easy to others.

Here the video: 

https://youtu.be/F57SC1WOODg

Thanks for your job.

Best regards.

ykri, 2017-01-02:
Hi @FernandoGarcia !

Great tutorial. 

It was also a chance for me to go through your [Freduino](https://github.com/FernandoGarcia/Ferduino_with_webcontrol_beta) code and try to figure out how you did make it work with MQTT. 
I run a similar project with my aquarium controller and although I manage to pub/sub with the El-Client using @tve 's example, after about 10' the transmission stops and I get erroneous payloads.
I cannot understand if it is that the SYNC fails, connection between Arduino and ESP is lost, or what!!!
I also tried your suggestion in [el-client #30](https://github.com/jeelabs/el-client/issues/30) to synchronise before publish. Still after almost exactly 10' the MQTT fails.

Any ideas?

Regards,

Y.

FernandoGarcia, 2017-01-05:
Hi!

Give more details about the error and your wiring maybe I can help you.

Try disable debug log in ESP-link web interface.

You can try the baud rate at 9600 too.

Best regards.

ykri, 2017-01-06:
Hi @FernandoGarcia !

An ESP-01 is connected to the RxTx of the Arduino Mega. I tried 115000 and 9600 as you suggested and debug log to OFF.
I run Mosquito in Linux and subscribe with NODE-RED.

When I start with MCU RESET this is the debug log output:
>414278> SLIP: start or end **len=64** inpkt=1
414278> cmdParsePacket: cmd=11 argc=5 value=0
414278> cmdExec: Dispatching cmd=MQTT_PUB
414278> MQTT: MQTTCMD_Publish topic=/Aquarium/Temperature1, **data_len=5,** qos=0, retain=0
414279> MQTT: Publish, topic: "/Aquarium/Temperature1", length: 31
414279> MQTT: Send type=PUBLISH id=0000 **len=31**
414394> SLIP: start or end len=0 inpkt=1

From NODE-RED the complete msg is: 
>Object{ "topic": "/Aquarium/Temperature1", **"payload": "26.50"**, "qos": 0, "retain": false, "_topic": "/Aquarium/Temperature1", "_msgid": "315e531a.2a253c" }

But after about 10'-11':
>179407> SLIP: start or end **len=60** inpkt=1
179407> cmdParsePacket: cmd=11 argc=5 value=0
179407> cmdExec: Dispatching cmd=MQTT_PUB
179407> MQTT: MQTTCMD_Publish topic=/Aquarium/Temperature1, **data_len=1**, qos=0, retain=0
179407> MQTT: Publish, topic: "/Aquarium/Temperature1", length: 27
179408> MQTT: Send type=PUBLISH id=0000 **len=27**
179523> SLIP: start or end len=0 inpkt=1

In NODE-RED now the msg is:
>Object{ "topic": "/Aquarium/Temperature1", **"payload": { "type": "Buffer", "data": [ 205 ] }**, "qos": 0, "retain": false, "_topic": "/Aquarium/Temperature1", "_msgid": "7a5c383c.283c08" }

I see that the payload from a value of "26.50" becomes { "type": "Buffer", "data": [ 205 ] }.

Thanks,

Y.

ykri, 2017-01-06:
OK I might have spotted the problem just after I explained it in the previous post.
It seems that my SRAM is exhausted because I use this function to convert float values to string before publish them:

>char * FtoS(float Value) {
  String vString = String(Value);
  int length = vString.length() + 1; // +1 allows room for trailing NULL
  char *result = malloc(length * sizeof(char)); // the sizeof here isn't really
                                                // necessary for char, but for
                                                // portability...
  vString.toCharArray(result, length);
  return result;
} 

Any suggestions of a proper FtoS function?

Thanks,

Y.

FernandoGarcia, 2017-01-06:
How about that?

char buffer[5];
String s = dtostrf(floatValue, 2, 5, buffer);

ykri, 2017-01-06:
That's it Fernando!!!

Thanks !!!

closed 2022-04-14 by FernandoGarcia