jeelabs/esp-link #233: How to flash to ESP8266 Module (WRL-13678)

rutes22, 2016-12-17:
I'm flashing from linux with the command "esptool --port /dev/ttyUSB0 --baud 460800 write_flash -fs 8m -ff 80m 0x00000 boot_v1.5.bin 0x1000 user1.bin 0xFE000 blank.bin"

Specs of the module :
Memory/Speed Features 80MHz 
64KB instruction RAM 96KB data 
RAM 64K boot ROM
1MB* Flash Memory

I have 3x 1mb modules. I erase the flash memories with esptool before install. When flashed with same command one of them works fine, second spams fatal error at UART, third works for a few seconds and reboots with watchdog reset. Is there anything I'm missing?

tve, 2016-12-18:
Please refer to the flashing instructions and don't forget to also flash the init data.
https://github.com/jeelabs/esp-link/blob/master/FLASHING.md

yigibihe, 2016-12-19:
I believe there is a mistake in that documantation. There is no "esp_init_data_default" in esp-link-v2.2.3.tgz. The correct way for flashing v.2.2.3 is in its own release notes, but it does not have any info on esp-01 1MB modules I own.

tve, 2016-12-19:
If you're flashing 2.2.3 then there indeed is no esp_init_data_default. The current readme has the numbers for a 1MB flash, 0xfe000 for blank.bin is correct.

yigibihe, 2016-12-20:
`[aaa@aaa tmp]$ curl -L https://github.com/jeelabs/esp-link/releases/download/v2.2.3/esp-link-v2.2.3.tgz | \
>     tar xzf -
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   596    0   596    0     0    877      0 --:--:-- --:--:-- --:--:--   877
100  448k  100  448k    0     0   136k      0  0:00:03  0:00:03 --:--:--  187k
[aaa@aaa tmp]$ cd esp-link-v2.2.3
[aaa@aaa esp-link-v2.2.3]$ ls
avrflash  blank.bin  boot_v1.5.bin  user1.bin  user2.bin  wiflash
[aaa@aaa esp-link-v2.2.3]$ `

Am I missing something?

yigibihe, 2016-12-20:
And " -fs 8m -ff 80m " is also correct given that module is 80MHz and 1MB?

tve, 2016-12-20:
yup

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve