jeelabs/esp-link #234: mqtt-link does not subscribe or print out data

shahabmusic, 2016-12-18:
my mqtt-link connects to mqtt server successfuly however when I publish something using another client, my mqtt-link does not print anything out via serial. It still prints keep alive details via serial though.

I tried to enable SLIP but still nothing get printed via serial once someone publish something.
So it is either esp-link not subscribing to the topic or there is some other unknown problem.
Could someone help me with this please?

tve, 2016-12-18:
esp-link does not magically subscribe to any topics, your sketch running in the attached uC needs to do that.  Please look at the mqtt example in the el-client repo: https://github.com/jeelabs/el-client/blob/master/ELClient/examples/mqtt/mqtt.ino

shahabmusic, 2016-12-18:
thank you. but I need to use the esp-link as a stand alone MCU without any attached uC. How can I approach this please?
in mqtt_client.c and inside mqttConnectedCb function, I added MQTT_Subsribe(client, "#", 0);.
But now whenever I publish TESTPAYLOAD into Topic1/Topic2, serial output is like:

152057> MQTT: Recv type=PUBLISH id=0000 len=28; Pend type=NULL id=00
152057> MQTT: Recv PUBLISH qos=0 Topic1/Topic2TESTPAYLOAD

I believe it should print out something like this:
MQTT Client: Received topic: Topic1/Topic2, data: TESTPAYLOAD

tve, 2016-12-19:
You're looking at the debug printout and it doesn't format the data portion correctly. You would have to fix that or somehow generate a proper subscription. But in the latter case it would wrap the message in a SLIP packet.
Overall, esp-link was not designed to subscribe to topics on its own and pass the resulting messages to something attached to the serial port.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve