jeelabs/esp-link #235: LoadStoreAlignmentCause

yigibihe, 2016-12-20:
Hello,

I flashed esp-link-v2.2.3. to my esp01 1MB module with following command. GPI02 was high and GPI01 was low.

 esptool --port /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -ff 80m 0x00000 boot_v1.5.bin 0x1000 user1.bin 0xFE000 blank.bin

I used esptool erase_flash to clean the memory before flashing. I get the following error and module resets.

>  15373> GET scan: cgiData=0 noAps=0
>  15379> GET scan: cgiData=1 noAps=0
>  15381> HTTP GET /wifi/scan: 200, 15ms, h=20976
>  15567> Fatal exception 9(LoadStoreAlignmentCause):
>  15567> epc1=0x4020ccca, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000035, depc=0x00000000
> 
>  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
> 
> wdt reset
> load 0x40100000, len 1856, room 16 
> tail 0
> chksum 0x63
> load 0x3ffe8000, len 776, room 8 
> tail 0
> chksum 0x02
> load 0x3ffe8310, len 552, room 8 
> tail 0
> chksum 0x79
> csum 0x79
> 
> 2nd boot version : 1.5
>   SPI Speed      : 80MHz
>   SPI Mode       : QIO
>   SPI Flash Size & Map: 8Mbit(512KB+512KB)
> jump to run user1 @ 1000
> 
> ����8�����r����8        ���r����8�����lrr��9l�Ğ����br�b������b�#�n��#�n��`llX��
>                                                                                
>                                                                                
>    129> 
>    129> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62
>    129> Flash config restore ok
>    129> CONN led=0
>    131> Wifi init, mode=AP+STA
>

tve, 2016-12-20:
clearly that's a bug :-) Did you use the user1.bin/user2.bin from the release zip file or did you compile from source? I hope I have the map file somewhere so I can see where the exception happened.

yigibihe, 2016-12-21:
Hello,

I burned the 2.2.3 from releases page with the command: 
`esptool --port /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -ff 80m  0x00000 boot_v1.5.bin 0x1000 user1.bin 0xFC000 0xFE000 blank.bin`

I deleted the flash with esp tool before writing.

esptool flash_id:
```
Manufacturer: e0
Device: 4014
```
Module: https://www.sparkfun.com/products/13678

If there anything else I can do please let me know. This firmware is great, I would love to help. 

Another interesting point I would like to mention, after few restarts sometimes it gets better and works fine for a while. Error code also changes occasionally.(UART debug gist [https://gist.github.com/yigibihe/4a323afec8bf9c9b1068088b433ae19f] )

tve, 2016-12-21:
Have you tried the latest version of esp-link? It would be easier for me to fix a bug there, if it's there too.

yigibihe, 2016-12-28:
I recently got contacted by the seller, I have received an ESP from a faulty batch. New modules arrived today and esp link v2.2.3 worked fine.

tve, 2017-01-01:
Phew! Thanks for reporting!

closed 2017-01-01 by tve