jeelabs/esp-link #238: Password to Web-Side

nikes, 2017-01-03:
I not found info, how set password to web-server side. But if share web side to net its so bad, anyone can connect to module.
Mb add in new update login & password to accept web-side?
P.S. or how/where i can add this auth method?

FernandoGarcia, 2017-01-06:
Hi!

I think there no way by default, you should edit the firmware.

Look for @dannybackx changes it's something like this:

```
os_strcpy(user, "admin");
os_strcpy(pass, "s3cr3t");
```

https://github.com/jeelabs/el-client/pull/25

Best regards.

dannybackx, 2017-01-06:
This is not in the main repository yet, it's work in progress in https://github.com/fuzzball03/esp-link.git/branches/testing/esp-link/main.c .

nikes, 2017-01-06:
Nice job! Its really good funcs for this module.

closed 2017-01-06 by nikes