jeelabs/esp-link #239: How to set baudrate from Client?

igittigitt, 2017-01-04:


igittigitt, 2017-01-06:
As i found in serial/serbridge.c there must be some kind of NVT implementation (RFC 2217). Can you provide some examples for this? I played around a bit, but not success :-(

Tried to set the baudrate with this sequence:
  FF FA 2C 01 00 00 01 2C FF F0

As seen in RFC document:
 IAC SB COM-PORT-OPTION SET-BAUD <value(4)> IAC SE

# IAC = 0xFF
# SB  = 0xFA
# COM-PORT-OPTION = 0x2C
# SET-BAUD = 0x01
# SE = 0xF0

tve, 2017-01-06:
The setting of the baud rate should work, as you describe, but I have to admit I'm not sure I've actually tested the baud rate setting specifically. I tested the other ones for sure 'cause I use them, but I may have missed the baud rate one.
This being said, there is a peculiarity to all this, which is that the client MUST send IAC and WILL or DO as the first two characters on the connection for the telnet escapes to be enabled. If the client doesn't, the telnet escapes are not enabled.

tve, 2017-01-06:
BTW, the debug log should print stuff about telnet, such as `os_printf("telnet mode\n");` when it's enabled on a connection and `os_printf("telnet mode\n");` when setting the baud rate.

igittigitt, 2017-01-06:
You're right. First, one need to send a "WILL" command first, as you said:
`FF FB 2C`
next, sending the baudrate (here 300 Baud) command:
`FF FA 2C 01 00 00 01 2C FF F0`
This will produce the desired log output and result in change of baud!
BUT: there is an issue with it, tough. If i send the command again, the rates will mix up in 2558 Baud or such. I think the problem is here, in serial/serbridge.c line 177:
` tn_baud |= ((uint32_t)c) << (24-8*tn_baudCnt);`
Because 'tn_baud' is not initialized and global:
`static uint32_t tn_baud; // shared across all sockets, thus possible race condition`
So simply adding a baud init somewhere, e.g. in line 112, could do the job:
` case SetBaud: state = TN_setBaud; tn_baudCnt = 0; tn_baud=0; break;`

tve, 2017-01-06:
Ugh, thanks for flagging the issue. I'll apply your fix!

igittigitt, 2017-01-06:
Great! Thank you for the fast support. Is there an official documentation (wiki) to tell about the NVT feature and give examples how to use it? If any help is needed here, i would be proud to help.

igittigitt, 2017-01-12:
After reading the code, i am pretty shure that there is no need to send IAC WILL cmd bevore any setup cmds. Also, there is no need to set COM params in any order, like stated in RFC2217.

And also, there may be problem if a virtual COM driver is not capable of NVT commands and the serial data contains 0xFF 0xFF, because this only results in a single 0xFF !

Well, i am still  issing some userland documentation about the Telnet command features. Where abd how could this bin done? Is there a "official" homepage/wiki for this project?

igittigitt, 2017-01-14:
I've tried to setup esp-link with [HW-Group Virtual COM driver](http://new.hwg.cz/files/download/sw/version/hw-vsp3-single_3-1-2.exe). It supports some of RFC2217 Telnet commands. It refuse to work because esp-link expects `IAC WILL COM-PORT-OPTION` as the first bytes after a Telnet connection is established. Otherwise it will fallback to transparent bridge mode.
Well, for some reason HW-Group's driver does not send this but begin with `FF FA 2C 01 00 00 00 00 FF F0` to query the current baudrate of the terminal.
For now i did not test other drivers, like com2tcp or such, how they behave.
Just two questions about this:
1.) Would'nt it be better to enable/disable Telnet commands in the Web-GUI of esp-link, instead of "magically" inspect and decide by incomming datastream? I'm not shure if Telnet commands must always identify itself first and not later in the stream. Normally, if a device supports Telnet commands it must handle escaped 0xFF's.
2.) It's not clean, but to support such faulty drivers, i would like to have option 1.) with a dropdown-list which driver to support, e.g. "RFC2271-Support" having optins like "Disabled", "Enabled", "HW-Group VCP"

tve, 2017-06-17:
Ooops, I never saw your further comments. I agree with you, PR welcome ;-)

zalexua, 2022-09-01:
Let me mention that that is another driver which potentially might be used 
https://github.com/rchovan/npreal2

uzi18, 2022-09-07:
@zalexua not maintained anymore.