jeelabs/esp-link #241: Setting up swapped uart with pull down resistor

noob999, 2017-01-07:
Hi ! First thank you for your excellent work. I get everything up and running using a wemos D1 but i dont managed to get it work with swapped uart configuartion. Do you have some schematics where to place the pulldown resistor ? If i place it to GPIO 15 it will boot but uc communication is not working. Thanks for your help.

jeco1964, 2017-01-13:
Hi. You are right. A resistor >3k, <10k between GPIO15 and ground should do the work.
but it does not.
something is wrong

noob999, 2017-01-13:
I think i found out whats wrong. I am using a voltage shifter for RX an TX and this shifter is using a Pullup for the 3,3 volt side wich causes PIN 15 to be high. So i think i have to use a different shifter but i don't know wich. Or i have to do some schematics wich diables the pullup as long as the ESP is booting. But i really have no Idea how to do it and my ideas are right.

tve, 2017-06-18:
closing 'cause nothing has moved here in a long time.

closed 2017-06-18 by tve