jeelabs/esp-link #243: Create BREAK signal using NVT in Telnet session

igittigitt, 2017-01-11:
Hello,

for my special project with esp-link, i need to be able to create a BREAK signal on the serial side of the ESP. The break is created by bringing TX up for a defined time, in my case 25ms, followed by a specified time TX low bevore sending any further bytes.
The serial standard (as described here https://www.cmrr.umn.edu/~strupp/serial.html#modem) call this TCSBRK, but there is also a way to pull TX up and down, to produce a break in custom length.
The signals could be produced via NVT (see RFC2217 http://www.hw-group.com/support/nvt/rfc2217.txt).

So whats missing in esp-link is to expand the NVT support by the needed commands, an change TX output line by reprogramming it to GPIO and back to UART mode after it. Currently i try to implement this in my github checkout.

How would you prefer doing the TX pulse? Is there something i can already use? The LED code seems a good starting point...

igittigitt, 2017-01-12:
I am working on an extension of RFC2217 command-set "IAC SB COM-PORT-OPTION SET-CONTROL <value> IAC SE" by the values:
5 = Set BREAK State ON
6 = Set BREAK State OFF
to set and reset the TX line to produce the desired BREAK signal.

NVT compatible virtual COM drivers could use this to implement tty_ioctl() with TCSBRK, TIOCSBRK and TIOCCBRK like described here http://man7.org/linux/man-pages/man4/tty_ioctl.4.html

tve, 2017-06-18:
Supported in v3.2, thanks!

closed 2017-06-18 by tve