jeelabs/esp-link #246: QUESTION: Is there a way to do a soft reset through cURL, HTTP or Telnet?

rjblake, 2017-01-14:
I have a PHP script that logs data from my boiler using an ESP with ESP-Link. On occasion, I need to reset this remotely. I'd like to be able to do this in the PHP script (after x errors, reset ESP), but can't get my head around how I can 'emulate'  the 'Reset esp-link' button or invoke ESP.reset() or AT+RST. Is this possible?

igittigitt, 2017-01-14:
There are several ways to reset esp-link remotely. One is to simply call the reset-URL using native PHP function: `file_get_contents("http://esp-link.fritz.box/log/reset");`

rjblake, 2017-01-18:
I'll need to look into that. For some reason it does not seem to be working in my script. I've tried from a browser window with http://My_ESP_IPAddress/log/reset and it doesn't seem to work either (well, if I look at the ESP debug, it doesn't seem to 'restart'

igittigitt, 2017-01-18:
The webgui of esp-link itself uses this method. So if you can issue a restart by clicking in webgui, it should also work from the script.

rjblake, 2017-01-18:
From browser, it wasn't working, or as far as I can see not (the debug log isn't reset and counter keeps incrementing). However, when I call from inside my PHP it does the job perfectly. Many thanks.

tve, 2017-06-18:
looks like this is resolved.

closed 2017-06-18 by tve