jeelabs/esp-link #247: ESP-01 only blue LED fast flashing after reboot, no more WiFi

igittigitt, 2017-01-15:
After playing around with Telnet commands, i managed to brick my esp-link somehow. I've tried to change baudrate, send some bytes and change baud back. After that esp-link crashes, baudrate was wired setting and the device does not respond anymore.
After resetting the device, red LED is on and blue LED is flashing forever. Connected HTerm to see what is going on, and see this looping:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000
```

Even after erasing the whole flash (by writing 1MB stub file at pos 0x0000) and write the software back again, this issue is still the same!

igittigitt, 2017-01-15:
Well, found a workaroud in the net, because many people also bricked their devices and bring them back to live by flashing the genuine Firmware ("v0.9.2.2 AT Firmware.bin") first and after that esp-link again.
This worked for me also. It seems that the bootloader is going to get corrupted somehow. Maybe there is a bug in esp-link which caused this (writing to wrong flash address).

leodesigner, 2018-06-03:
Hi,
Seems I have the same issue, but it was after flashing latest version.

esptool.py --port /dev/cu.wchusbserial1410  --baud 230400 write_flash -fs 32m -ff 80m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin

I have a boot loop with the following message:
```
rf cal sector: 128
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.1.0(116b762) compiled @ May  5 2017 16:08:55
phy ver: 1134_0, pp ver: 10.2

FLASH chk=0x0000 crc=0x0000 full_sz=1024 sz=600 chip_sz=0
mqtt_host is ''
Fatal exception 0(IllegalInstructionCause):
�B*r��P���K�.�dmXK�~�`�hXHꡁ���r�,T�����Ĩ�␃��~A���n((�8x��P�݁��*��t�,�@j|��@��*���n((�8x�vA$Z[!�((�+k��J(�+��*��*���n(Al")z�~�PY�r~���X�����.�␄r�+�����+����lA�v�����m %j��␄���N�dA#�x��v��␄X�[p�h�{�␂�R�k~P~��hmm�P����rf cal sector: 128
freq trace enable 0
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.1.0(116b762) compiled @ May  5 2017 16:08:55
phy ver: 1134_0, pp ver: 10.2

FLASH chk=0x0000 crc=0x0000 full_sz=1024 sz=600 chip_sz=0
mqtt_host is ''
Fatal exception 0(IllegalInstructionCause):
```

Then I am flashed stable esp-link-v3.0.14:
esptool.py --port /dev/cu.wchusbserial1410  --baud 230400 write_flash -fs 32m -ff 80m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin

```
rf cal sector: 128
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.0.0(5a875ba) compiled @ Aug  9 2016 15:12:27
phy ver: 1055, pp ver: 10.2

FLASH chk=0x0000 crc=0x0000 full_sz=1024 sz=600 chip_sz=0
mqtt_host is ''
Fatal exception 0(IllegalInstructionCause):
epc1=0x4000e25d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
�@*r��P���K�.�dr�%@��.�Ѫ�Ko�X���*�.������P�,���*����nzAn Jz�*�j�H~t�R����@j+��*��*���nzAf Jz�~�,�H~t�R���␂��Ĩ�K�~�(��lzAl )�m*�t�8r� �ǁ�@+��.��m�+��P�(�V�jt������x (�V�jP�����n (�V��x��v��␄X�[p�h�{`���Z�k~P~��hmp�P����rf cal sector: 128
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.0.0(5a875ba) compiled @ Aug  9 2016 15:12:27
phy ver: 1055, pp ver: 10.2

FLASH chk=0x0000 crc=0x0000 full_sz=1024 sz=600 chip_sz=0
mqtt_host is ''
Fatal exception 0(IllegalInstructionCause):
epc1=0x4000e25d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
�@*-��(���I�t�dr�-�z+P~�-H������*�T�����Ĩ���{~AB��n((�8x�vA,v%#�DP�+k,����P�-@��*���nzAl*)�vA,X%#�DP�+k챩����I�~�(��n((�8x�nA,t�8r*�*�X�����,�␄r�+��� (�j�v�(������@�`A�t�(������`A)�KPYx�+�␄-�[p�h�{`�␂�X��n,�Am|�␛p�P�����rf cal sector: 128
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 2.0.0(5a875ba) compiled @ Aug  9 2016 15:12:27
phy ver: 1055, pp ver: 10.2

FLASH chk=0x0000 crc=0x0000 full_sz=1024 sz=600 chip_sz=0
mqtt_host is ''
Fatal exception 0(IllegalInstructionCause):
epc1=0x4000e25d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
Looking now for v0.9.2.2 AT Firmware.bin to unbrick device.

leodesigner, 2018-06-03:
Flashed old 2.2.3 version:
```
rf[112] : 00
rf[113] : 00
rf[114] : 01

SDK ver: 1.5.4(baaeaebb) compiled @ May 17 2016 19:23:54
phy ver: 972, pp ver: 10.1

FLASH chk=0x0000 crc=0x0000 full_sz=1024 sz=596 chip_sz=0
mqtt_host is ''
Fatal exception 0(IllegalInstructionCause):
epc1=0x4000e25d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
```
At the same time when I am flashing this board with Platformio (Arduino framework),
the board is functional and flashed firmware is working. Really strange.

leodesigner, 2018-06-03:
Git it fixed, seems I had FlashConfig from previous flashing of eep-link. So I had to erase_flash and now  I am able to run firmware again.

ximen, 2019-03-05:
Confirm issue. Works well while newly flashed, but starts crashing after saving new STA WiFi settings on next reboot.
Tested on latest 3.2.47 and ESP01.

NetForces, 2020-01-23:
I have just experienced this, bricking 2 devices... Unfortunatly uploading the `v0.9.2.2 AT Firmware.bin` and then re-uploading the esp-link firmware does not solve the issue for me.

Here is the log that keeps on going:

```
2nd boot version : 1.7(5d6f877)
SPI Speed : 80MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22
```

I also tried erasing the ESP8266 flash and re-uploading the `v0.9.2.2 AT Firmware.bin` firmware and then the esp-link firmware, same thing...

Any other thing I can try to unbrick these and restart using the esp-link firmware ?

tve, 2020-01-23:
There's no such thing as "bricked" esp8266's. Sometimes the HW may be bad and a reflash may expose that, e.g. if the flash chip has bad blocks, often the power supply is insufficient. If these are OK, start by doing a flash erase with esptool 'cause it looks like some of the blocks used by the bootloader are bad, e.g. the RF calibration stuff.

NetForces, 2020-01-23:
@tve Yeah I know it's not bricked cause I can upload a simple code via the Arduino IDE (or Plarformio) and it runs it. But as soon as I flash the esp-link firmware it goes into a reboot loop.

As for erasing I sed the following command: `esptool.py --port /dev/tty.usbserial-1410 --baud 115200 erase_flash`

I just tried flashing it again but added the `-e` and `--verify` switches.

```
{23:21}[2.5.0]~/Desktop/esp-link-v3.2.47-g9c6530d ➭ esptool.py --port /dev/tty.usbserial-1410 --baud 115200 write_flash --flash_freq 80m --flash_mode dio -e --verify --flash_size 4MB 0x0000 boot_v1.7.bin 0x1000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin
esptool.py v2.8
Serial port /dev/tty.usbserial-1410
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:fd:8c:50
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.8s
Flash params set to 0x024f
Compressed 4080 bytes to 2936...
Wrote 4080 bytes (2936 compressed) at 0x00000000 in 0.3 seconds (effective 112.1 kbit/s)...
Hash of data verified.
Compressed 344708 bytes to 253827...
Wrote 344708 bytes (253827 compressed) at 0x00001000 in 24.7 seconds (effective 111.8 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 75...
Wrote 128 bytes (75 compressed) at 0x000fc000 in 0.0 seconds (effective 74.8 kbit/s)...
Hash of data verified.
Compressed 4096 bytes to 26...
Wrote 4096 bytes (26 compressed) at 0x000fe000 in 0.0 seconds (effective 3649.5 kbit/s)...
Hash of data verified.

Leaving...
Verifying just-written flash...
(This option is deprecated, flash contents are now always read back after flashing.)
Flash params set to 0x024f
Verifying 0xff0 (4080) bytes @ 0x00000000 in flash against boot_v1.7.bin...
-- verify OK (digest matched)
Verifying 0x54284 (344708) bytes @ 0x00001000 in flash against user1.bin...
-- verify OK (digest matched)
Verifying 0x80 (128) bytes @ 0x000fc000 in flash against esp_init_data_default.bin...
-- verify OK (digest matched)
Verifying 0x1000 (4096) bytes @ 0x000fe000 in flash against blank.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...
```

So the readback was good. 

I'm also powering from a DC-DC power supply capable of providing up to 18W so it's not a power supply issue.

At least one of these board worked with esp-link before I re-flashed it.

tve, 2020-01-23:
Hmmm, not sure. Did you try 40Mhz flash speed? And you're sure you have a 4MB part?

NetForces, 2020-01-23:
I bit late tonight to check the chip marking. I'll check that tomorrow, but 99% sure. Yesterday I flashed the v3.0.14-g963ffbb version to that same board and it was working. Earlier today I flashed v3.2.47-g9c6530d and it was working good as well. I wanted to document the procedure so I went head and erase the flash and re-flash v3.2.47-g9c6530d and that's when it started going bad.

I did try 40MHz with no difference.

NetForces, 2020-01-23:
So I'm back on my feet... Looks like I have taken in note the wrong offsets. I'm using the RobotDyn Mega+EST8266 32Mb so the following is the proper flash commands:

```
curl -L http://s3.voneicken.com/esp-link/esp-link-v3.2.47-g9c6530d.tgz | tar xvf -
cd esp-link-v3.2.47-g9c6530d
esptool.py --port /dev/tty.usbserial-40120 --baud 115200 write_flash --flash_freq 80m --flash_mode dio --flash_size 4MB 0x0000 boot_v1.7.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
```

uzi18, 2020-01-23:
so init data default  and blank. bin was at wrong addresses

so it was good point from tve to check flash size