jeelabs/esp-link #248: Hardware configuration schematic

seanlano, 2017-01-16:
I just installed esp-link onto an ESP-12 module for the first time - so exciting! Thanks for the great work. 
One thing I struggled with was getting the initial hardware setup correct. (It didn't help I was using a breakout board that seemingly doesn't work with serial flashing, so I made my own circuit for it instead). 

I've made a couple of schematics in Fritzing for the description in [FLASHING.md](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md) - it helped me a lot to have it drawn out while setting it up. Is there any reason against including a schematic on the FLASHING.md page? If not then I can tidying my drawing up and submit it.

tve, 2017-06-18:
Oops, I overlooked your message :-(. If you still want to provide the schematics, I'd include them. Thanks & sorry.

tonertiffi, 2020-04-29:
Hi Seanlano,
If you still have those schematics for the ESP01 connected to an Arduino MEGA i would really appreciate it if you could mail them to me, as i am struggling to understand on the ESP01 as the coment stands " GPIO2: optionally connect green LED to 3.3V (indicates wifi status)" i would assume connect GPIO TO 3.3V .That seems wrong to me , or maby it means gpio2 ,insert a green led then wire the one leg to 3.3v, every circuit i have looked at uses a resistor.Please help if you can. My email is hassio.scott@outlook,com,thanking you in advance.

seanlano, 2020-05-01:
Hi @tonertiffi ,
Unfortunately I don't have the schematic anymore, I think I sketched them on some paper which has long since gone. But looking at it again now I think the comment about GPIO2 for for the ESP-01 module should be something similar to the ESP-12 module, something like this:

> either a 1k-10k pull-up resistor to 3.3v or a green "conn" LED via a 1k-2.2k resistor to 3.3V (indicates wifi status)

The GPIO2 pin is used for setting the boot mode, as you note it needs a resistor to 3.3V - or an LED with a lower-value resistor.

Also I suppose I'll close this issue, since I don't have the schematics anymore.

closed 2020-05-01 by seanlano