jeelabs/esp-link #250: Does esp-link support serial flash STM32F103xx ?

blackadmin, 2017-01-17:
Hi,
I am successfully able to flash NXP LPC824 using esp-link, but unable to flash STM32F103xx using the same pin configuration.

**esp-link v2.2.3**
GPIO13: connect to TX of microcontroller
GPIO15: connect to RX of microcontroller and using 10K pull-down
GPIO1: connect to RESET of microcontroller
GPIO3: connect to ISP of LPC/ARM microcontroller (**BOOT0 in case of STM32F103xx**)

Using NXP dev-board : https://developer.mbed.org/platforms/LPCXpresso824-MAX/
Using STM dev-board : https://developer.mbed.org/platforms/ST-Nucleo-F103RB/

Please help. :(

tve, 2017-01-17:
If I understand correctly you need to use 8bit even parity for the stm32, did you do that? That's either using the web UI or the telnet escape sequences.

blackadmin, 2017-01-20:
Tried parity 8E1, still no success. :(
Changed esp-link version to latest but no success.

I am able to flash STM using Serial adapter.

Below are the debug logs from esp-link. It seems like esp-link keeps on resetting microcontroller.
```

406384> Telnet: reset gpio1
407135> MQTT: Send type=PINGREQ id=0000 len=2
407460> MQTT: Recv type=PINGRESP id=0000 len=2; Pend type=NULL id=00
408385> Telnet: reset gpio1
409044> HTTP GET /console/text: 200, 6ms, h=14544
410396> Telnet: reset gpio1
412442> Telnet: reset gpio1
413069> HTTP GET /console/text: 200, 7ms, h=14544
414592> Telnet: reset gpio1
416140> HTTP GET /console/text: 200, 7ms, h=14544
416392> Telnet: reset gpio1
```

blackadmin, 2017-01-20:
I have flashed a program in STM32 using serial adapter which prints "Hello" on UART1, and then afterwards I can see "Hello" in microcontroller console of esp-link using **FMT : 8N1 (not 8E1)**

blackadmin, 2017-01-26:
I think why I got data on esp-link console using **8N1** format is because me using [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32) for programming my STM32 MCU and when using **Serial** for printing, its documented in Arduino that by default parity **8N1** will be used for **Serial**.

But still no success with flashing. :(

One more query, will **el-client** work on my STM32 MCU if me using [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32) ?  

Thank you for such a good library.. :)

bitdump, 2017-01-28:
I have flash stm32F103 using uart bootloader before. I need to change some code in esp-link. The first one is the uart need to use the 8E1 format instead of 8N1. The second thing is that, if I remember correctly, the ISP boot pin value is inverted in stm32 compare to AVR. In stm32, in order to enter boot loader mode, stm32 require "boot0=1". For AVR, the "ISP=0" before enter bootloader mode.

The first change I have submit to esp-link and it is merged.
The second change I just did a quick and dirty way to flip the ISP value
because I don't program AVR any more. It has been a while, I don't
use bootloader mode to flash stm32 any more either.

The code change for ISP pin is at "serial/serbridge.c", go search for "mcu_isp_pin".

blackadmin, 2017-01-28:
@bitdump thank you very much for reply, I am gonna give it a try.

PriankaAg, 2017-01-30:
Hi , 
@bitdump
"I have flash stm32F103 using uart bootloader before. I need to change some code in esp-link. The first one is the uart need to use the 8E1 format instead of 8N1. The second thing is that, if I remember correctly, the ISP boot pin value is inverted in stm32 compare to AVR. In stm32, in order to enter boot loader mode, stm32 require "boot0=1". For AVR, the "ISP=0" before enter bootloader mode.

The first change I have submit to esp-link and it is merged.
The second change I just did a quick and dirty way to flip the ISP value
because I don't program AVR any more. It has been a while, I don't
use bootloader mode to flash stm32 any more either.

The code change for ISP pin is at "serial/serbridge.c", go search for "mcu_isp_pin"."

I am also in same stituation, tried this solution file is megered and change in mcu_isp_pin  
 Here 
if (startPGM) {
#ifdef SERBR_DBG
    os_printf("MCU Reset=gpio%d ISP=gpio%d\n", mcu_reset_pin, mcu_isp_pin);
    os_delay_us(2*1000L); // time for os_printf to happen
#endif
    // send reset to arduino/ARM, send "ISP" signal for the duration of the programming
    if (mcu_reset_pin >= 0) GPIO_OUTPUT_SET(mcu_reset_pin, 0);
    os_delay_us(100L);
    if (mcu_isp_pin >= 0) GPIO_OUTPUT_SET(mcu_isp_pin, 1);
    os_delay_us(2000L);
    if (mcu_reset_pin >= 0) GPIO_OUTPUT_SET(mcu_reset_pin, 1);
    //os_delay_us(100L);
    //if (mcu_isp_pin >= 0) GPIO_OUTPUT_SET(mcu_isp_pin, 1);
    os_delay_us(1000L); // wait a millisecond before writing to the UART below
    conn->conn_mode = cmPGM;
    slip_disabled++; // disable SLIP so it doesn't interfere with flashing
#ifdef SKIP_AT_RESET
    serledFlash(50); // short blink on serial LED
    return;
#endif
  }

bitdump, 2017-02-02:
@PriankaAg, Your change looks good.

Esp chip reset the micro-controller, that looks like a firmware flashing problem.
Have to able to verify that, after you issue the command to put stm32 into
bootloader mode. You can actually flashing the stm32 using the uart port on
stm32?

That will confirm you have actually put stm32 into bootloader mode successfully.

How do you flashing stm32 from the wifi socket provided by esp-link?
Existing tools using uart. How do you change to socket interface?
A lot of timing assumption is different when using wifi. wifi latency is
a lot longer than local UART. I recall I have to change to the code for
that as well on the bootloader flashing tool. I am using stm32loader.py

hetii, 2017-05-23:
Hello :)
I wrote a patch for stm32flash that allow use it with esp-link.
Currently its in dirty stage so no automatic baudrate/mode setting, but will try to add it in few days.

So here are my  steps to flash stm32f030:

0. Clone and build my version of stm32flash from: https://bitbucket.org/hetii/stm32flash
1. Pull boot0 high.
2. Reset stm32 by low pulse on NRST.
3. Run stm32 to flash your software.
4. Change back boot0 to low and reset uC.

RESULT:
```
./stm32flash

 192.168.0.10 -v -w ../soft.bin 
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Socket created
Connected

Interface serial_net: 
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0444 (STM32F03xx4/6)
- RAM        : Up to 4KiB  (2048b reserved by bootloader)
- Flash      : Up to 32KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 3KiB
Write to memory
Erasing memory
Wrote and verified address 0x08000220 (100.00%) Done.
```

blackadmin, 2017-06-12:
@hetii thankyou very much, will give it a try. :)

PriankaAg, 2017-06-14:
@heti thank you so much.

blackadmin, 2017-06-20:
@hetii I tried but had no success :(  I have STM32F103

OUTPUT:
```
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Socket created
Connected

Interface serial_net: 
```

and it remains there only.

sadransh, 2017-08-17:
@blackadmin I had the same Issue the problem is inside the esp-link settings. just get into settings go to uC Console and set the baud as 115200 and the Fmt on 8E1. it will be OK then.

sadransh, 2017-08-17:
@blackadmin  I saw you have already tested that. try this : 
1.download Hercules http://new.hwg.cz/files/download/sw/version/hercules_3-2-8.exe 
2.go to TCP client mode connect with 192.168.4.1 port 23 ( when connected to UART to micro)

3.use boxes below the app. write 7f and check the hex box. then reset the micro by key and press send you should be able to get y character
 then send 00ff if you Fmt is correct you should get a string from the controller.

blackadmin, 2017-08-21:
@sadransh I will give it a try and revert, thanx :)

blackadmin, 2017-08-25:
@sadransh I tried your method and yes it worked, please mention in your comment to set micro BOOT0=1. Thank you very much :).
And also flashing worked. Thanx to @hetii 

```
./stm32flash -b 115200 -m 8e1 -w firmware.bin 192.168.0.141
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Socket created
Connected

Interface serial_net:
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (STM32F10xxx Medium-density)
- RAM        : Up to 20KiB  (512b reserved by bootloader)
- Flash      : Up to 128KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote address 0x08002124 (100.00%) Done.
```

What I observed is that 
1) Sometimes I have to reset ESP and after few seconds reset micro and try flashing then it gets flash, other wise it gets stuck.
2)  Using command **./stm32flash** with argument `-v` gives error, without `-v` program gets flash and works fine.
```
./stm32flash -v -b 115200 -m 8e1 -w firmware.bin 192.168.0.141
stm32flash 0.5

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Socket created
Connected

Interface serial_net:
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (STM32F10xxx Medium-density)
- RAM        : Up to 20KiB  (512b reserved by bootloader)
- Flash      : Up to 128KiB (size first sector: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote and verified address 0x08000300 (0.94%) Failed to read memory at address 0x08000300
```

blackadmin, 2017-08-25:
Also I am trying some way to flash micro using ESP without any serial bridge, means like ESP downloads the firmware from cloud and flashes it to micro, but dont know how to start with it :(, I am checking the @hetii stm32flash code and if I can embed flashing code into esp-link, any help I will be very thankful. :)

blackadmin, 2017-10-05:
Successfully done it, thanks to @sadransh and @hetii https://bitbucket.org/hetii/stm32flash

hm-wu, 2018-01-26:
@sadransh 
In your comment you mentioned 
> 2.go to TCP client mode connect with 192.168.4.1 port 23 ( when connected to UART to micro)
SET boot0 =1

Do you mean "set boot0 = 1" in esp-link? If yes, how can I manually set a pin in esp-link?

sadransh, 2018-01-28:
yes i mean manully set boot0 pin of controller to 1 
acctually there are to ways to do that 
1. there are 2 protocols(not sure) in esp-link firmware learn one of them and send commands throw uart to it to switch pins 
2. change codes inside the uart recieve callback of the esp-link and build it once more ,

sadransh, 2018-01-28:
@tve  & @blackadmin  I think you should close this

closed 2018-01-28 by blackadmin

uzi18, 2018-01-28:
I will chceck this issue,
maybe it is possible to integrate more automatic transision to pgm mode of stm32.
Have some stm32f103 and need to try.

Flodee, 2018-10-24:
hey thank you all for the informations here. I can flash the maple mini overt the air (ota) through esp-link now. BUT only if i bring it into flash mode by clicking the buttons.. How can i set the boot0/isp pin automatically when uploading. I dont get what @sandransh is proposing ...

ibrahimballoul, 2025-04-13:
Hi, i know i like 6 years late, but dude @hetii your stm32flash with esplink support is just amazing, you have made my life much easier, i can program my stm32 without touching it.....thanks Alot

uzi18, 2025-04-14:
@ibrahimballoul what about boot0 configuration? please add screenshots of esp-link configuration and version

ibrahimballoul, 2025-04-14:
hi @uzi18, i am using STM32 Blackpill, but your lucky i have a bluep

[image]

ill laying around, i tested it also with the esplink working also, For Boot0 i removed the yellow jumper and Connected esplink GPIO2(ISP-PIN) to the middle pin, i have tested esplink latest release 3.2.47, but for some reason i didn't know i wasn't able to connect to the 2323 programming port while 23 port is open, maybe it's my fault i only tried a couple of times, while browsing and reading through issues, in this one https://github.com/jeelabs/esp-link/issues/386 i found this repo https://github.com/leodesigner/esp-link.git, working great for me. ps: if you used this version don't forget to invert the ISP pin logic from the esplink home page

ibrahimballoul, 2025-04-14:
Also another thing, if your are on Windows i haven't tested @hetii  https://bitbucket.org/hetii/stm32flash version of stm32flash tool, i am using stm32flash 0.7 release , i am using com2com with com2tcp to create the virtual Serial COM port,  on linux i am using hetii's  stm32flasher version working great.

uzi18, 2025-04-14:
You can check my latest build

uzi18, 2025-04-14:
@ibrahimballoul please add also screenshots of esp-link configuration for other users :)

ibrahimballoul, 2025-04-14:
sorry for telling you that PS earlier :), and thank you for your contributions to the ESP-LINK project....i will test your build in a couple of days and post my findings......, sorry for my question but what do you mean exactly by esp-link configuration?

ibrahimballoul, 2025-04-14:
hi, i got a bit enthusiastic and tested your build, Working!, i can now use port2323, thanks....the tests i made were on windows using stm32flash tool 0,7...

I used Espressif flash download tool "i am a poor windows guy :) using Ubuntu for ROS", and usb2ttl converter with FTDI232R chip to flash the binary files, i used this configuration to flash the esp-01, it has 8Mbit flash memory chip.

boot_v1.7.bin @ 0x00000
blank.bin @ 0xFE000
esp_init_data_default.bin @ 0xFC000
user1.bin @ 0x1000
SPI Speed : 40MHz , SPI Mode : DOUT

I am using esp-01 GPIO0 for the MCU reset PIN ,  GPIO2 for the MCU BOOT0 "aka ISP" pin
PS: I am using a Schmitt-Trigger inverter to invert the ISP pin Logic, Stm32F401 and F103 i tested with needs 3,3V to enter bootloader mode.

Wiring Diagram is similar to the picture above, with addition to the inverting Schmitt-Trigger gate.
GPIO2----- >input----->inverted output----->BOOT0

The baud rate on both mcu console web page and com2tcp must be similar, and Even Parity also must be used...at least for the MCU's i have tested...

 @uzi18  if there anything i can add also please tell me

stm32flash output:
C:\ST\STM32CubeIDE_1.18.0\STM32CubeIDE>stm32flash -g 0x8000000 -b 115200 -w "E:\Microcontrollers Programing Codes\LOCAL_CUBEIDE_WORKSPACE\tests\Debug\tests.bin" -v COM1 
stm32flash 0.7

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Size         : 11116
Interface serial_w32: 115200 8E1
Version      : 0x31
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0433 (STM32F401xD(E))
- RAM        : Up to 96KiB  (12288b reserved by bootloader)
- Flash      : Up to 512KiB (size first sector: 1x16384)
- Option bytes  : 16b
- System memory : 30KiB
Write to memory
Erasing memory

Wrote and verified address 0x08000100 (2.30%) 
Wrote and verified address 0x08000200 (4.61%) 
Wrote and verified address 0x08000300 (6.91%) 
Wrote and verified address 0x08000400 (9.21%) 
Wrote and verified address 0x08000500 (11.51%) 
Wrote and verified address 0x08000600 (13.82%) 
Wrote and verified address 0x08000700 (16.12%) 
Wrote and verified address 0x08000800 (18.42%) 
Wrote and verified address 0x08000900 (20.73%) 
Wrote and verified address 0x08000a00 (23.03%) 
Wrote and verified address 0x08000b00 (25.33%) 
Wrote and verified address 0x08000c00 (27.64%) 
Wrote and verified address 0x08000d00 (29.94%) 
Wrote and verified address 0x08000e00 (32.24%) 
Wrote and verified address 0x08000f00 (34.54%) 
Wrote and verified address 0x08001000 (36.85%) 
Wrote and verified address 0x08001100 (39.15%) 
Wrote and verified address 0x08001200 (41.45%) 
Wrote and verified address 0x08001300 (43.76%) 
Wrote and verified address 0x08001400 (46.06%) 
Wrote and verified address 0x08001500 (48.36%) 
Wrote and verified address 0x08001600 (50.67%) 
Wrote and verified address 0x08001700 (52.97%) 
Wrote and verified address 0x08001800 (55.27%) 
Wrote and verified address 0x08001900 (57.57%) 
Wrote and verified address 0x08001a00 (59.88%) 
Wrote and verified address 0x08001b00 (62.18%) 
Wrote and verified address 0x08001c00 (64.48%) 
Wrote and verified address 0x08001d00 (66.79%) 
Wrote and verified address 0x08001e00 (69.09%) 
Wrote and verified address 0x08001f00 (71.39%) 
Wrote and verified address 0x08002000 (73.70%) 
Wrote and verified address 0x08002100 (76.00%) 
Wrote and verified address 0x08002200 (78.30%) 
Wrote and verified address 0x08002300 (80.60%) 
Wrote and verified address 0x08002400 (82.91%) 
Wrote and verified address 0x08002500 (85.21%) 
Wrote and verified address 0x08002600 (87.51%) 
Wrote and verified address 0x08002700 (89.82%) 
Wrote and verified address 0x08002800 (92.12%) 
Wrote and verified address 0x08002900 (94.42%) 
Wrote and verified address 0x08002a00 (96.73%) 
Wrote and verified address 0x08002b00 (99.03%) 
Wrote and verified address 0x08002b6c (100.00%) Done.

Starting execution at address 0x08000000... done.

uzi18, 2025-04-15:
@ibrahimballoul Thanks, in fact screenshots of esp-link website - configuration + console should be ok
Nice idea with inverter, not sure if we implemented it for esp-link, maybe some patches waiting, will check it later.

ibrahimballoul, 2025-04-15:
[image]
[image]

Here are the configurations i used with my esplink

uzi18, 2025-04-15:
Thank you, will think about inversion option for isp/reset

ibrahimballoul, 2025-04-15:
Take a look to this issue https://github.com/jeelabs/esp-link/issues/386, i have tested the build mentioned in it and it's working, it has ISP inversion only, reset is just the same as other builds, also it has the option to select higher baud rates in the Mcu console page among other modifications he made, i haven't inspected all of them thoroughly yet....however, he mentioned something about a pull request....i don't know if he made that....