jeelabs/esp-link #251: avrdude -> esp-link -> arduino

swilson86, 2017-01-18:
On a Windows 10 laptop I cannot upload a new arduino sketch using avrdude. I have compiled the sketch with arduino IDE to get a hex file. The destination is to an arduino mini pro connected to an ESP-01 chip loaded and configured with esp-link. Here is the code (and error) I see at a command prompt:
-------
avrdude -DV -patmega328p -Pnet:192.168.0.25:23 -carduino -b57600 -U flash:w:Blink.hex -C avrdude.conf
avrdude: ser_drain(): read error: The parameter is incorrect.
avrdude: ser_drain(): read error: The parameter is incorrect.
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x53
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x74
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x65
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x70
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x31
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x36
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x0a
avrdude: ser_drain(): read error: The parameter is incorrect.
avrdude done.  Thank you.
-------
Things I tried: Changing baud rate, various versions of avrdude, set admin priviledge for the command prompt, the letter :i just after the .hex in the command above. Things I observe: I can use reset on the uConsole and the arduino does reset. When I hit the enter key after entering the long command above I do see the one blink of the blue LED on the ESP-01 also one blink on LED blinks on the arduino.

Anyone have ideas? This is driving me crazy every day for 2 weeks.

1nfrag, 2017-01-19:
I've had problems too. Ive set baud rate to 115200 and flash direct to serial from arduino sketch.. No manual conversion necessary as I use a virtual com port usr-vcom I thinks its called. To get this to work I need to manually reset it at exactly the right moment using the esplink webpage. I find pinging the esp IP address seems to help make the timing required more consistent for some reason. It can still take me 5 tries but it does work like that for me.

Anyone else have any input?

1nfrag, 2017-01-19:
Oh and BTW thanks for this awesome firmware. Very happy with it so far!

FernandoGarcia, 2017-01-19:
Hi!
If your target is only update the sketch remotely, follow my suggestion here: https://github.com/jeelabs/esp-link/issues/87

Best regards.

swilson86, 2017-01-20:
1nfrag, brilliant suggestion with USR-VCOM which I downloaded today. However, it gave me the same results as avrdude. When is "exactly the right moment" to hit the reset on on the esplink page? Your solution is miles ahead of avrdude at a command prompt, fooling with hex files and lengthy commands.

swilson86, 2017-01-20:
Fernando, thanks but nothing on #87 seemed to help out so far. An ESP-12 arrived today (as opposed to my ESP-01). I am going to start swapping out things until I find the source of this pesky problem. I am still suspicious that I have overlooked something silly... especially that Home > Pin Assignment > ISP/Flash section which is set to disable. It gives me pin conflicts if I try to enable it with an ESP-01.

FernandoGarcia, 2017-01-20:
If you'll use an ESP-12 you can try this library to burn the sketch via ISP instead UART.

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP

1nfrag, 2017-01-20:
@FernandoGarcia my target wasnt only to remotely update the sketch. I use the MQTT parts of ESPLINK which was what I used the ESP-01 for in the first place. Using it to remotely flash was an important feature but one that I can work around if required.

@swilson86 the exact right time for me was (while pinging the ESP to make sure it had stable wireless connection while flashing) just after it finished compiling obviously but hard to get exactly the right time. I turned on compiling debugging to give me a better indicator on how far along compiling was.... anyway if I got the result you had it was usually after it was trying to upload and i hit the reset (so therefore too late). If I reset to early usually the result was a hanging repeated attempted to upload the sketch.

I'm stilling waiting on my ESP-12 to see if the results are better. If it works for you better please keep us updated!

swilson86, 2017-03-28:
Thanks for all the ideas above on uploading Arduino code to the ESP-01 chip. The ultimate solution which works perfectly is to use avr-link. Download avr-link from https://gitlab.com/bc547-playground/avr-link  For  those finding this note the procedure is... a) Compile your Arduino sketch in the normal way with the IDE program Arduino provides. Find the hex file which ends up in a /temp folder (look at the verbose compile info to determine where it placed the file). b) Copy the ino.hex file to a safe place ie: c:/ArduinoHex c) Place the avr-link software in the same folder. d) At a command prompt copy that hex file to Arduino over the air (OTA) using avr-link. The command is similar to avr-link flash --esp 192.168.1.35  --file SketchBlah.ino.hex -v e) Interesting side effect: Find all your esp-link modules with the command avr-link esp discover -n 192.168.1.0/24 -v

closed 2017-05-25 by swilson86

blogwebpl, 2021-10-16:
I have this error only on windows:

avrdude: ser_drain(): read error: The parameter is incorrect.
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x53

on macos everything works (avrdude6.3):
% avrdude -carduino -pm8 -Pnet:192.168.1.149:23
ioctl("TIOCMGET"): Device not configured
ioctl("TIOCMGET"): Device not configured

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9307 (probably m8a)

avrdude: safemode: Fuses OK (E:FF, H:00, L:00)
ioctl("TIOCMGET"): Device not configured

avrdude done.  Thank you.

blogwebpl, 2021-10-16:
esp-link firmware in esp01 connected to atmega8 rx tx and reset. using avrdude. on mac ok on windows error.