jeelabs/esp-link #252: Atmega328 flashing

Ulli2k, 2017-01-22:
I tried to flash a connected Atmega328p to an NodeMCU.
It works with the following command
`avrdude -q -V -p atmega328p -D -c arduino -b 19200 -P net:esp:23 -U flash:w:RadioNetwork.hex:i`
but just after setting the baudrate to 19200 in the esp-link webinterface.

Two questions:
* do I always need to change the baudrate in the webinterface for flashing and after flashing set it back to the baudrate I am using in my sketch?
* I tried using avrflash but it did not work. I always get the following error: 
  `Error checking sync: No response in state init(0) @19200 baud`
   What is the issue on that?

Thanks a lot for your support in advance.

njanik, 2017-01-23:
Thanks for reporting this issue, you've helped me a lot :)

I've just made a workaround. I use avrflash script (that come with esp-link) to flash my atmega.
I just simply add a curl request in this script (before try getting the sync) to set the correct baud rate. (the same request that is made manually when you change the baud rate via the uC console page).

`setBaud=`curl -s -w -XPOST "http://$hostname/console/baud?rate=115200"``

Ulli2k, 2017-01-24:
Welcome, sounds good!
I will try to add your command in my built flash procedure... thanks.

But all this is a workaround. I thought the avrflash script should manage all that?

Ulli2k, 2017-02-05:
One additional question. Is it possible to always reset the Atmega which is connected to the esp-link when a new connection on port 23 is established?

tve, 2017-02-06:
use port 2323 to get the reset

Ulli2k, 2017-02-07:
Wow it was really that simple. Thanks!
But it looks like that I have a big delay (up to 5s) in the communication over Wifi. Is that normal?

Ulli2k, 2017-02-11:
Oh I recognized that using port 2323 does not restart the Atmega....is there a configuration I need to do?

yesyesuk, 2017-02-11:
under Pin assignment -> Reset on the main page you need to select the pin that is connected to the reset input on your Arduino.

tve, 2017-06-18:
closing, seems to be resolved

closed 2017-06-18 by tve