jeelabs/esp-link #253: flash the atmega: maximum sketch size with esplink?

njanik, 2017-01-23:
Hello

Does the size of the sketch can be problematic when I want to use esplink to flash my distant atmega?

After a lot of testing, here’s my conclusion:

Tested with 3 differents ESP01, with 3 differents atmega328 and differents power sources.

I compil my sketch with arduino IDE and flash the atmega via esplink using the script “avrflash” that come with the esplink release.

It’s working very well with all my differents ESP01 or atmega328 when I upload the “blink” example.
But when I try to flash my own sketch, it’s working only **ONE time**.
To be able to do it again, I have to reflash the “blink” example on my atmega (I use a real arduino uno via USB for this purpose). After that, I can flash my sketch again but still only ONE time.

The error is:
`Error checking sync: No response in state init(0) @115200 baud
`
I supose that the difference between the blink example and my sketch is the size.
When I compile my sketch via the arduino IDE, the IDE is telling me:

```
Sketch uses 18,834 bytes (58%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,721 bytes (84%) of dynamic memory, leaving 327 bytes for local variables. Maximum is 2,048 bytes.
Low memory available, stability problems may occur.
```

Do you think the problem is related?

Thank you, and thank you for this wonderful library.

njanik, 2017-01-23:
In fact, I've just read the issue #252 , and that's it! My sketch set the serial baud rate to 9600. I have to put it back to 115200 in the uC console page to be able to flash it again. 
I did not understand that I have to change it manually.

tve, 2017-06-18:
Looks like you were hitting a bug I found today. Should be fixed in the next alpha release (>v3.1.44). 5d0e51f2c1aa96280f1d37252e04d4c5e32df2ea

closed 2017-06-18 by tve