jeelabs/esp-link #254: Support of 802.1x (WPA2-Enterprise  using EAP or PEAP)

igittigitt, 2017-01-24:
Hello,
currently, esp-link only offers connections to "Open" (unsecured) or Password-protected (PSK) WLANs. To use it in more secure environments like enterprise Wifi's, it would need have 802.1x support also.

WPA2-Enterprise support is there since SDK 1.5 ([see this releasenote](http://bbs.espressif.com/viewtopic.php?f=46&t=1442)), so it should be possible.

gnif, 2017-05-22:
+1

I can't use this in my environment. I have some ESP8266 modules on the lan using EAP-PEAP with MSCHAPv2, so it is indeed possible and should be a minor change to add support.

gnif, 2017-05-22:
I have had a go at hacking this into the EAP 2.1 branch, the issue is there is too much memory overhead by esp-link. EAP requires 26KiB of RAM, not including the space that would be required if one were to provide the root CA for certificate verification.

tve, 2017-06-18:
I'm afraid that as you describe EAP is not realistic unless the memory usage is somehow drastically slimmed down.