jeelabs/esp-link #258: Lack of security, allowing anyone with network access to mess with the device

dreamlayers, 2017-02-02:
This firmware allows anyone with network access to mess with the device.

For the most part this is not a big deal. Changing the settings and messing with the attached microcontroller would just be a nuisance. This would only be a problem if the attached microcontroller was controlling something important where interference can cause serious problems.

The worst part is that anyone with network access can flash a new firmware to the ESP8266. This means they could create a rogue device on the network.

tve, 2017-02-03:
As you mentioned this is somewhat a duplicate of #19. I'm not planning to add security without also going HTTPS and that's not simple if you actually deal with certs properly and has nasty memory consumption implications.

tve, 2017-06-18:
Closing as dup of #19

closed 2017-06-18 by tve