jeelabs/esp-link #259: Ping delays and WiFi disconnection when writing large blocks to port 23 with low baud rate

dreamlayers, 2017-02-03:
The ESP-01 module running the latest esp-link is configured for 600 baud. It's connected to my LED sign, but that probably doesn't matter.

I'm running an executable which connects to port 23 and then sends a large block of data, and then immediately running ping after that. Up to 100 bytes is totally okay. As size increases beyond 100, the first ping latency increases. Pings after the first have normal latency. At block size 221, the first ping has 2 seconds latency. Blocks of that size and larger cause esp-link to disassociate from the WiFi network where it it was running in station mode. I see an esp-link network now, but cannot connect to that.

So, it seems like there's some buffer of a fixed size, and when I fill that buffer everything waits until more characters can be put into that buffer. I guess that waiting is in uart_tx_one_char() in the SDK, though that will put up to 126 characters in the buffer.

The problem seems to be that writing to the UART, which will block if its transmit buffer is full, is happening inside the ESP8266 connection callback, serbridgeRecvCb(). I suppose the answer is to buffer data and use espconn_recv_hold() and espconn_recv_unhold() to prevent that buffer from overflowing.

dreamlayers, 2017-02-04:
It's possible to work around this from the client side. `setsockopt(fd, SOL_SOCKET, SO_SNDBUF,  &size, sizeof(int))` can make it work indefinitely with size 100. There is a disadvantage though, as there are irregular short pauses in transmission, and not constant character after character at 600 baud.  This causes jitter in LED sign scrolling. After adding `setsockopt(fd, IPPROTO_TCP, TCP_NODELAY, &flag, sizeof(int))` with flag set to 1, the scrolling is smooth even with buffer size 0, and it works indefinitely. This was tested in Ubuntu 16.10 with Linux 4.8.0-34-generic. So, it seems I have a good workaround for this bug.

tve, 2017-06-18:
Yup. The real solution would be to make uart TX interrupt-driven. That's on my to-do list... I'm going to close this ticket in the meantime.

closed 2017-06-18 by tve