jeelabs/esp-link #260: udp.begin fails

gui77aume, 2017-02-05:
Hello, first of all, thanks for this nice project !
Sorry, I'm not sure this is the place for this kind of question/issue:

Currently trying to use esp-link with el-client on arduino Uno wifi, I'm stuck with an error:
Running the example UDP.ino,
I get :
`UDP begin failed: -1029`

after running:

`

    int err = udp.begin(udpServer, udpPort, SOCKET_UDP, udpCb);

    if (err < 0) {

    Serial.print(F("UDP begin failed: "));

    Serial.println(err); 
...
`

The versions I use are:
Esp Link version : master of 01/28/2017
esp SDK version : esp_iot_sdk_v2.0.0.p1
EL client version : master of 01/26/2017

Debug log on the esp side:

`992357> SOCKET Setup listener flag
992394> SOCKET #0: Setup host=192.168.0.12 port=5000 
992451> SOCKET #0: Create connection to ip 192.168.0.12:5000
992516> cmdResponse: cmd=2 val=0 argc=0
992571> SOCKET #0: setup finished`

What am I doing wrong ? 
What is the -1029 error code meaning ?

Thanks !

Guillaume

gui77aume, 2017-02-09:
It works after an esp-link reset while the controler is running the udp sketch.

tve, 2017-06-18:
@beegee-tokyo any thoughts?

beegee-tokyo, 2017-06-20:
@tve and @gui77aume 
Strange, the log file indicates that the ESP side successfully setup the socket connection, but then the return value should be the socket slot used (0 to 3).
In case the socket setup fails the return code would be -1 to -7, but not -1029.

-1 SOCKET Setup parse command failure: (cmdGetArgc(&req) != 3)
-2 SOCKET Setup parse command failure: hostname longer than 128 characters
-3 SOCKET Setup failed to alloc memory for socket_host
-4 SOCKET Setup parse command failure: (cmdPopArg(&req, socket_host, len))
-5 SOCKET Setup parse command failure: cannot get port
-6 SOCKET Setup parse command failure: cannot get mode
-7 SOCKET #%d: Setup failed to alloc memory

My guess would be that the error is created within the communication part between the ESP and the Arduino, but I do not have a unit available to debug, because my Uno WiFi is used in one of my projects (using 1 UDP socket and 1 TCP socket). It runs since months without any problems.

If it always works after an esp-link reset then it might be a temporary solution to initiate such a reset in the setup() of the Arduino.

@tve
I actually added into el-client and esp-link the possibility to programmatically reset the ESP and the Arduino. See attached files (ELClientCmd.cpp and ELClientCmd.h for el-client, handlers.c and cmd.h for esp-link). Sorry, I never pushed these changes to your repo, they are only in the beegee-flavor branch of my fork.

[cmd.h.txt](https://github.com/jeelabs/esp-link/files/1087810/cmd.h.txt)
[ELClientCmd.cpp.txt](https://github.com/jeelabs/esp-link/files/1087811/ELClientCmd.cpp.txt)
[ELClientCmd.h.txt](https://github.com/jeelabs/esp-link/files/1087812/ELClientCmd.h.txt)
[handlers.c.txt](https://github.com/jeelabs/esp-link/files/1087813/handlers.c.txt)