jeelabs/esp-link #262: 3.0.x feature: custom web pages

uzi18, 2017-02-06:
it is possible to add only 1 web page

uzi18, 2017-03-10:
@tve why not use spiffs like other projects?

tve, 2017-06-18:
That's an idea... I didn't implement the web page feature (and I don't use it)...

yesnoj, 2018-02-16:
Ok, but how remove web pages once uploaded!?

uzi18, 2018-02-16:
upload empty html file

yesnoj, 2018-02-16:
uzi18 thanks for reply, but i can't understand how...i've the SimpleLed webserver loaded, how can i remove this page?!

overisberg, 2018-07-09:
yesnoj just press "Submit" in the Web Server page without selecting any file to remove the file you previously uploaded.
It took me a while to figure this out ;)

overisberg, 2018-07-09:
uzi18 you can select multiple files and upload them in one go.
If you select just one file it will replace all the previous files you uploaded.
If you select no files all the files you previously uploaded will be removed.