jeelabs/esp-link #263: UDP only sometimes works

p0llard, 2017-02-09:
Using Release 3.0.14 Alpha on Arduino UNO WiFi with EL-Client Release v3.0.alpha0.

I have a rather strange issue when running [the EL-Client demonstration code for UDP sockets](https://github.com/jeelabs/el-client/blob/master/ELClient/examples/udp/udp.ino). Essentially, it only sometimes works; quite a lot of the time the serial console will say that it is sending the packet, but never confirms that it has been sent - if this behavior occurs, then it will not stop until after the arduino has been reset several times. Resetting esp-link through the online interface fixes the problem immediately, but a sub

Upon the successful sending of a packet, I would expect `socketclient_sent_cb` to be called, which sends a command over serial to the Arduino, triggering the callback within the EL-Client library. It seems that `socketclient_sent_cb` is never being called, as when a packet fails to send we don't see the expected output in the log (see bottom).

After 'extensive debugging' (resetting the Arduino several times and looking at the esp-link online debug log), it seems that if the sockets used are #0 or #1, then there usually isn't a problem - normally the packets fail to send if sockets #2 or #3 are used... ...except when it's the otherway around, which happens entirely unpredictably.

In fact, adding an additional UDP socket usually results in only two of them working, normally the ones which are assigned #0 and #1, except sometimes very rarely #1 and #2 will work (or #2 and #3, or #3 and #0). I say usually as one time it worked fine with three sockets, but I haven't been able to reproduce this behaviour.

Has anyone else experienced this unusual behaviour? I'm going to try and get my hands on a replacement board, so I'll close the issue if that solves it.

Packet Successfully Sent Log:
```
...
2238> cmdParsePacket: cmd=41 argc=2 value=0
2239> cmdExec: Dispatching cmd=SOCKET_SEND
2239> SOCKET #0: send dataLen=50 socketData=Message from your Arduino Uno WiFi over UDP socket
2239> SOCKET #0: Create connection to ip 192.168.0.102:5000
2239> SOCKET #0: sending 50 bytes: Message from your Arduino Uno WiFi over UDP socket
2240> SOCKET #0: Sent
2240> cmdResponse: cmd=3 val=942 argc=3
...
```

Packet Send Failure Log:
```
...
702055> cmdParsePacket: cmd=41 argc=2 value=1
702055> cmdExec: Dispatching cmd=SOCKET_SEND
702055> SOCKET #1: send dataLen=50 socketData=Message from your Arduino Uno WiFi over UDP socket����
702055> SOCKET #1: Create connection to ip 192.168.0.102:5000
702055> SOCKET #1: sending 50 bytes: Message from your Arduino Uno WiFi over UDP socket����
...
```

Notice in the second log the absence of the `Sent` and `cmdResponse` entries.

gui77aume, 2017-02-11:
I'm dealing with the same kind of behaviour: sometime it works, sometimes the callback function isn't called. What is your pin assignment ?

tve, 2017-06-18:
@beegee-tokyo any thoughts?

beegee-tokyo, 2017-06-20:
@p0llard and @tve 
No idea atm why it runs only with 2 UDP sockets. it should work with 4 sockets, UDP and TCP sockets counted together, as they share the code. 

Are there any other sockets/connections in use (TCP, MQTT, REST, ...)???

In other projects with the ESP8266 (pure ESP projects, no Arduino board involved), I got problems if I try to open too many sockets/connections, e.g. HTTP, FTP, TCP server, TCP client and UDP client altogether. The ESP8266 dropped some connections, it seems there is some limitation in the ESP sources regarding the number of sockets that can be used in parallel.
I found this: [LIMITS ON NUMBER OF SOCKETS ?](http://www.esp8266.com/viewtopic.php?f=8&t=10033) check the response from dannybackx.

I do not have a unit available to debug, because my Uno WiFi is used in one of my projects (using 1 UDP socket and 1 TCP socket). It runs since months without any problems.

leodesigner, 2018-06-16:
Hi,
I made some fixes to the socket.c, this should fix exactly this issue with socket reassignment.
The problem here is the esp8266 is trying to open again the same UDP socket again and to to reuse it.
Due this esp_send return error -12. So I made the changes to allow to reuse already opened UDP sockets. Also I am added ability to use hostname instead of IP for UDP sockets.
You may test my branch at https://github.com/leodesigner/esp-link/. Also there is precompiled binary
in firmware directory.