jeelabs/esp-link #264: error: ip_addr.h: No such file or directory

shahabmusic, 2017-02-10:
I used to compile the old version fine. As soon as I pull the new update, started getting this error.
I renamed the folders to something else and downloaded the new esp-open-sdk and build it but I am still getting that error.
What could be the issue?

Thanks in advance

shahabmusic, 2017-02-10:
I just decided to create a new virtual machine and setup everything from scratch. New esp-open-sdk and new esp-tool. but still getting the same error.

tve, 2017-02-10:
is there a reason you're not using the docker image to build?
I have not pulled a fresh esp-open-sdk in a while, so something may have happened, or you're somehow building it differently. You are downloading the Espressif SDK 2.0 separately and applying the patch, right?

shahabmusic, 2017-02-10:
I don't know what is docket! I have seen it somewhere but I have never tried it.
I am building esp-open-sdk with standalone=n
No. I don't download SDK separately. esp-open-sdk apparently downloads it.

Thanks for the reply. I go away and do some research about docker and downloading SDK separately.

tve, 2017-02-10:
you have to download the SDK separately, the one included in esp-open-sdk has modifications that cause compilation failure. If you get the SDK from Espressif you have to also get the patch for 2.0.0 although ti will mostly work without the patch, I believe. You can also look into the .travis.yml file for the instructions for the automated travis-ci build.

shahabmusic, 2017-02-10:
wowww that docker image was amazing. made my life lot lot easier. I did research about it and eventually managed to compile and build esp-link so easy. No more virtual boxes, no more linux needed. I will try to upload the firmwares to the boards and see if the result is same.

Thank you very much for your time and very useful information.

closed 2017-02-10 by shahabmusic