jeelabs/esp-link #266: ESP8266 WiFi Shield unstable

TheNitek, 2017-02-10:
I got one of those awful ESP8266 WiFi shields (by WangTongze) for my Arduino. For it to make some sense I tried to get esp-link running on it.
I was able to flash it (3.0.14 Alpha and 2.2.3, both make the same problems), but unfortunately the webinterface is very unreliable: Sometimes it works fine, sometimes it loads forever. Most of the time the redirect to /home.html still works, but the webpage itself doesn't load. Sometimes the page even loads, but instead of the choose for the PINs, only a spinner is shown. If that's the case, telnet on 23 is possible, but doesn't show the UART (in other cases it works fine).
I was not able to identify any pattern but the problem occurs quite often and I don't have an idea how to debug it any further.

igittigitt, 2017-02-12:
I had such issue too, and was going crazy because of it. I also noticed the power LED (red one) pulsing irratically. Even a 1.000uF cap parallel to power supply does'nt made any change.
Then i decide to change from my China breadboard/jumpercables to more quality one and all issues where gone!

TheNitek, 2017-02-12:
Interesting! My power LED is bright and not pulsing but the STA and AP LEDs are glowing dimmed at the moment - Sometimes after a power on they'll glow in full brightness. Maybe that's a different symptom of a similar problem. But in my case it cannot be the cables, because I also see the problem, when the shield is plugged directly onto the UNO

TheNitek, 2017-02-12:
I did some more testing there doesn't seem to be a setup which doesn't have this problems. Does not seem to be the power supply

tve, 2017-06-18:
Did you ever resolve the issue?

TheNitek, 2017-06-18:
More or less: https://claus.bloggt.es/2017/02/26/arduino-esp8266-wifi-shield-elecshop-ml-by-wangtongze-comparison/
Seems to be depending on the version of the shield

tve, 2017-06-18:
interesting... I'm glad you got something working in the end!

closed 2017-06-18 by tve