jeelabs/esp-link #267: ESP-01 Upgrade Firmware page not loading

chinswain, 2017-02-11:
Hi,

I've just flashed an ESP-01 device with esp-link, all seems ok except for the firmware update page, it just sits on the loading animation:

[image]

The device was flashed on windows with the latest release:
esptool.py --port COM3 --baud 460800 write_flash -fs 4m -ff 40m 0x00000 boot_v1.6.bin 0x1000 user1.bin 0x7C000 esp_init_data_default.bin 0x7E000 blank.bin

tve, 2017-06-18:
Sorry, I didn't see your issue in time. Did it eventually come back? Or did you refresh the page? Or?

Badiboy, 2017-07-10:
Got the same issue all the time I visit the "Upgrade Firmware".
ESP-01, latest esp-link v3.2.47-g9c6530d.

_PS: I do not need this (AFAIK OTA firmware update is not applicable to ESP-01 with 512K memory), but the issue occurs. :)_

radiolee, 2018-02-06:
Got the same issue all the time too. 
ESP-01 with 1Mbit SPI Flash, latest esp-link v3.2.47-g9c6530d. Upload .bin just on the loading animation as chinswain's screenshoot.

uzi18, 2018-02-06:
I'm not sure if ota update reqly work for esp-01

radiolee, 2018-02-06:
My eps-01 module really own 1MBit SPI Flash...