jeelabs/esp-link #270: Uno Wifi

scallas, 2017-02-23:
Hi there
Is there a specific way/method to flash the Arduino Uno Wifi's ESP8266?
Im using Windows Vista.

Cheers
Schalk

astronomer80, 2017-02-25:
Hi, see this guide:
http://www.arduino.org/learning/tutorials/advanced-guides/how-to-change-the-firmware-on-an-arduino-uno-wifi
or this one
http://www.arduino.org/learning/tutorials/advanced-guides/arduino-uno-wifi-firmware-updater

With the Arduino Uno WiFi Develeper edition I think that the best way is to use the OTA. Note that using the USB2Serial version without the patch on the bootom of the board, the maximum baudrate is 9600.

scallas, 2017-02-25:
Thanks. Will try it. 

Schalk

scallas, 2017-02-25:
I tried it but am unable to select the esp-link firmware with the Uno Wifi updater tool.  Do I have to copy the firmware in a certain folder?

jandrassy, 2017-06-27:
Instructions in FLASHING.md are ok for the Uno Wifi dev. ed. Don't forget to prepare the board with EspRecovery sketch and bootloader mode.
(btw. Uno Wifi updater tool has the firmware list hardcoded.)

closed 2018-01-06 by scallas