jeelabs/esp-link #275: Extreme lag in serial data

rlesac, 2017-03-25:
Hi!
I'm using ESP-Link as a serial bridge for an Arduino Adalight project. 
Using serial to tcp software and ESP-Link I get over 250ms of lag compared to straight serial connection.
Using Windows 10 and HW VSP3 and com0com.

aaqibch, 2017-03-27:
Hi 
dont use HW VSP3 for creating virtual serial port. I was facing the same issue, try some other software for creating virtual serial port like TCP COM Bridge etc

rlesac, 2017-03-27:
Thanks!
I tried com2tcp with com0com, it was marginally faster, ~30ms.
Do you have a free alternative to TCP COM Bridge? I couldn't get their demo to work.

aaqibch, 2017-03-27:
you can also try NetBurner Virtual COMM Port Driver both worked for me :)

rlesac, 2017-03-27:
> you can also try NetBurner Virtual COMM Port Driver both worked for me :)
That one cut the lag in half, but it's still 100ms over serial.
How much lag do you get?

aaqibch, 2017-03-29:
Actually i didn't measure the lag i was just looking for faster bit rate while programming uC over wifi. I guess as esp-link itself is only about 63% efficient so there is always gonna be some lag.