jeelabs/esp-link #277: does not work web-server HELP

nsklital, 2017-04-03:
If you uncomment these lines. Then the LED on and off does not work
help.
ds18b20 temperature sensor pin 2

//sensor_inhouse.requestTemperatures();
    //tmp=sensor_inhouse.getTempCByIndex(0);

kod--------------

#include <ELClient.h>
#include <ELClientWebServer.h>

// flash LED on PIN 13
//#define LED_PIN 13
#define rl_ten 10
//---------------------------
#include <OneWire.h>
#include <DallasTemperature.h>

#define ONE_WIRE_BUS_1 2
#define term_power 4

OneWire oneWire_in(ONE_WIRE_BUS_1);

DallasTemperature sensor_inhouse(&oneWire_in);
float tmp=0;
//-----------------------------------------------------
// Initialize a connection to esp-link using the normal hardware serial port
//
// DEBUG is disasbled as
// - packet logging is slow and UART receive buffer can overrun (HTML form submission)
ELClient esp(&Serial);

// Initialize the Web-Server client
ELClientWebServer webServer(&esp);

void ledPageLoadAndRefreshCb(char * url)
{
  if( digitalRead(rl_ten) )
    webServer.setArgString(F("text"), F("LED is on"));
  else
    webServer.setArgString(F("text"), F("LED is off"));
}

void ledButtonPressCb(char * btnId)
{
  String id = btnId;
  if( id == F("btn_on") )
    digitalWrite(rl_ten, true);
  else if( id == F("btn_off") )
    digitalWrite(rl_ten, false);
}

// Callback made form esp-link to notify that it has just come out of a reset. This means we
// need to initialize it!
void resetCb(void) {
  Serial.println("EL-Client (re-)starting!");
  bool ok = false;
  do {
    ok = esp.Sync();      // sync up with esp-link, blocks for up to 2 seconds
    if (!ok) Serial.println("EL-Client sync failed!");
  } while(!ok);
  Serial.println("EL-Client synced!");
  
  webServer.setup();
}

void setup()
{
  Serial.begin(115200);
  //--------------
  pinMode(term_power, OUTPUT); 
    sensor_inhouse.begin();
    //--------------------------
  pinMode(rl_ten, OUTPUT);
  
  URLHandler *ledHandler = webServer.createURLHandler(F("/SimpleLED.html.json"));
  ledHandler->loadCb.attach(&ledPageLoadAndRefreshCb);
  ledHandler->refreshCb.attach(&ledPageLoadAndRefreshCb);
  ledHandler->buttonCb.attach(&ledButtonPressCb);

  esp.resetCb = resetCb;
  resetCb();
}

void loop()
{
  //--------
  digitalWrite(term_power, HIGH); 
    //sensor_inhouse.requestTemperatures();
    //tmp=sensor_inhouse.getTempCByIndex(0);
    //------------
  esp.Process();
}

tve, 2017-06-18:
I'm sorry, but at the moment I don't have time to reverse-engineer your code...

uzi18, 2018-02-16:
It is because sensor_inhouse.requestTemperatures(); blocks for 750ms. so esp-link goes out of sync.

add 2 lines at the end of setup():
```
sensor_inhouse.setWaitForConversion(false);
sensor_inhouse.requestTemperatures();

```
and change loop() to:

```
unsigned long czas = 0;
void loop()
{
  //--------
  digitalWrite(term_power, HIGH); 

    if (millis()-czas>1000) {
        tmp=sensor_inhouse.getTempCByIndex(0);
        sensor_inhouse.requestTemperatures();
        czas=millis();
    }
    //------------
    esp.Process();
} 

```