jeelabs/esp-link #278: Enhancement: snooping mode

sle118, 2017-04-03:
I am suggesting that esp-link implements a snoop mode to allow sniffing on serial protocol conversations. This would be extremely useful when debugging/analyzing conversations between devices without disrupting them.  Telnet would log data from one of the device's point of view with tags like TX/RX prefixing lines.

hexagon5un, 2017-04-06:
(I'm not related to the project other than being a happy user.)

Snooping one line is easy -- just attach an ESP-link RX pin to it and turn off the pullup.  Snooping two is left as an exercise to the motivated reader. 

Seriously -- the UART hardware TX line doesn't do receive natively, so this would have to be done in software.    This should be doable, but it's a different device altogether and a reasonable code project.  Throwing another ESP at the problem is probably the easy way out.

tve, 2017-06-02:
The esp8266 only has a single usable uart RX pin, so you would only be able to snoop one line, which is not what you're looking for, if I understand correctly.

closed 2017-06-02 by tve

sle118, 2017-06-02:
@tve i was thinking along the lines of soft serial, actually, using one of the very few pins of the esp8266