jeelabs/esp-link #279: Reconnect to network after router reset?

hexagon5un, 2017-04-13:
We are tormented by a flakey WiFi router that requires a power cycle once in a while. All devices on it need to be reset to reconnect, and I've programmed some of them to do so automatically.  

Is something like this doable with esp-link?  

I noticed that esp-link has a local-AP fallback mode when it can't (initially?) connect.  Rebooting every 30s (or whatever) upon lost connection would conflict with this behavior, but might also be a useful option for people who have and esp-link embedded in a device in a hard-to-reach corner of the basement.

tve, 2017-04-13:
I don't understand the issue. If you restart the AP then esp-link should reconnect OK. Does it not? What do you observe? Do you have debug logs showing what it's doing?
(All this is pretty much managed by the SDK, there is not a lot of code about this type of stuff in esp-link itself.)

hexagon5un, 2017-04-13:
Hmm... I'll have to look into this.  Two of my devices went down for a few days the last time we reset the router, but it's possilbe that it was a problem with my code.  Will have to set up a test network and try it out.

hexagon5un, 2017-04-20:
You're totally right.  It _was_ reconnecting, but user code (aka me being dumb) was keeping it from working.

Let's all just pretend this "issue" didn't happen?  :)

Thanks again for the very nice tool.

closed 2017-04-20 by hexagon5un