jeelabs/esp-link #281: Documentation suggestion: how to use serial converter/bridge once the wifi configuration is established

dpapathanasiou, 2017-04-14:
I've successfully completed the [hardware configuration](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md), [serial flashing](https://github.com/jeelabs/esp-link/blob/master/FLASHING.md#initial-serial-flashing), and [wifi configuration](https://github.com/jeelabs/esp-link/blob/master/WIFI-CONFIG.md) steps from the instructions, and I've got it connected to (and powered by) a Raspberry Pi without any other network interface.

So I was hoping to see the esp8266 chip available as a wlan adapter by running 'sudo iwconfig', but it's not present.

Then I saw this question -- https://github.com/jeelabs/esp-link/issues/274 -- which made me realize the intended usage is different from what I was expecting.

Are there more instructions on how to use the serial converter/bridge in this configuration?

I can connect to the Pi directly, with a monitor and keyboard, and I also have the ssh server enabled, for remote access.

I'm just not sure on what to do next, to establish connectivity.

tve, 2017-06-02:
to be honest, esp-link is not what you're looking for. the official "AT" firmware from Espressif is better suited for this type of use, I believe there are linux drivers available to interface to it.

dpapathanasiou, 2017-06-17:
Hi, and thanks for replying to this.
I came here by following [this article](http://pwiatrowski.com/technology/raspberry-pi-zero-esp8266-internet/) which specifically recommends esp-link for the original Pi Zero.
WRT using the AT firmware from Espressif instead, are there particular articles or references you could point me to?

tve, 2017-06-18:
I can't tell whether this project got completed: https://hackaday.io/project/8678-rpi-wifi but it's what you're looking for. Also http://hackaday.com/2015/12/09/raspberry-pi-wifi-through-sdio/