jeelabs/esp-link #285: How fast is the serial bridge

cambazz, 2017-05-01:
Hello,

I am building a ros based robot, and using the rosserial library, to get data from a robot and transmit it over with a 433mhz radio. However, it is not that fast, and not full-duplex, which causes problems.

Can I use the esp-link software to get a full duplex 115200 serial bridge? How about 230400bauds or faster?

Best regards,
C.

tve, 2017-05-01:
Yes, in that esp-link provides a full-duplex serial bridge at 115200 baud by default. However, if you're streaming at that rate without flow-control it's unlikely to work due to hiccups in the wifi. For example, if a TCP ACK is missed it's not uncommon for its retransmission to take several hundred milliseconds. Every 100ms requires ~1KB of buffer in esp-link and there's not a lot of RAM to go around... It shouldn't take long for you to try out and it will work but may not be long-term reliable.

cambazz, 2017-05-01:
I think that rosserial has some kind of flow control, but also ability to reestablish connection.

I am tried to build, but it failed with:

/bin/sh: 1: /home/can/Desktop/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: not found

is there a precompiled image, which I can just flash, or a prerequisites list for me to install before the makefile can work?

Best regards,

C.

tve, 2017-05-01:
https://github.com/jeelabs/esp-link/releases
There's also a docker image to build it: https://github.com/jeelabs/esp-link/blob/master/BUILDING.md#docker-linux-or-windows

closed 2017-05-25 by tve