jeelabs/esp-link #287: Unable to startup if baud set to 2400 or lower

zaddi, 2017-05-17:
After setting the baud rate to 2400 or lower, if the device is power cycled it fails to start-up properly and the ESP8266 has to be recovered by reprogramming the flash.  

This appears to be a repeating watchdog timeout due to the debug log on UART0 taking too long during start-up.  Disabling the UART debug log prevents the issue from occurring - perhaps the UART debug log should be disabled by default?

tve, 2017-05-17:
Interesting, sounds totally plausible. I recommend you compile your own version with debug disabled. I certainly don't want to make that the default 'cause I have to support the thing ;-).

zaddi, 2017-05-18:
Fair enough on not changing the default.  At least there's a thread now detailing the problem and easy workaround (ie disabling the UART debug log) to help anyone else with the issue.

closed 2017-05-18 by zaddi

vamposdecampos, 2017-05-18:
Perhaps one could kick the watchdog in-between debug messages?