jeelabs/esp-link #288: OLED display, software uart, sd card support and more.

hetii, 2017-05-24:
Hello dear :)

I think about few additional features for esp-link.

First is to add OLED support, second is a multiple sofware uart implementation that can use any gpio lefts.
Also, nice to have is sd card support where user can select firmware for destination mcu.
And finally more mcu should be supported, like stm32 or pic.

OLED support I made already and work on software uart.
But after reading project goals that esp-link is not swiss-army knife i`m not sure what I can add to it and what not.

So please feel free, and give me your opinion regarding my ideas.

In attachment I put status view from my oled:
[image]

uzi18, 2017-05-25:
I realy like idea to attach SD Card as user webpages or firmware storage
for slave mcu.

But now esp-link looks like abonded project, lots of patches are not merged
into upstream.
i.e. megaflash patches works quiet well.

tve, 2017-05-25:
It's not abandoned :-). I have a bunch of esp-links running 24/7 and forwarding mqtt data. I don't have a need for new features and I don't have much time to review and test lots of new stuff. Also, I prefer to keep esp-link focused on being a serial bridge with a web interface and not add a ton of other stuff. That being said, others have used it as a codebase to add their own fancy features, which is great.

tve, 2017-06-18:
I'm closing this for now, feel free to reopen.

closed 2017-06-18 by tve