jeelabs/esp-link #293: ESP-link disable errors and warnings on uart

wchpikus, 2017-06-02:
hi
Could i ask for option "disable" all warnings or info sending from server to UART?
When server work with wt32 bluethooth, to many garbage data reset module.
Only from tcp connection should be write to uart..

i got also problem with connect 2 clients to server, one of them only working ok.
Second recived info about problem with create tx bufor..
regards

tve, 2017-06-02:
> Could i ask for option "disable" all warnings or info sending from server to UART?

I'm not sure I understand what you are referring to. Which "server" do you mean?
You can disable the debug output on uart1 using the web UI.

>i got also problem with connect 2 clients to server, one of them only working ok.
>Second recived info about problem with create tx bufor..

Which version of esp-link are you using? Others have reported something similar and it seems it has to do with high data rates or poor wifi connectivity resulting in too much buffer space being required. Do you get specific error messages?

wchpikus, 2017-06-02:
Version 2.2.3.
Debug..i thinging about client connection e.g.
On uart i got info about this.
Info about error when incorrect data was inconning to server..on debug some like >stdin syntax error or somenthing..
All this message i have in putty and wt32 connected to sever..
I like to send and recive data only incomming from socket to uart.
Is it possible?

wchpikus, 2017-06-02:
I will try to capture this error tx buffer..

wchpikus, 2017-06-03:
ok, i got message e.g like this:

set
SET BT BDADDR 00:07:80:43:54:a3
SET BT NAME test aux
SET BT CLASS 200404
SET BT AUTH * 0000
SET BT IDENT BT:47 f000 5.0.2 Bluegiga iWRAP
SET BT LAP 9e8b33
SET BT PAGEMODE 4 2000 1
SET BT PAIRCOUNT 1
SET BT PAIR 44:78:3e:5b:09:8d 71520ac6827ea18e1c7d44e1a66c7d44
SET BT POWER 0 0 0
SET BT ROLE 0 f 7d00
SET BT SNIFF 0 20 1 8
SET BT SSP 3 0
SET BT MTU 667
SET CONTROL AUDIO INTERNAL / I2S I2S
SET CONTROL AUTOCALL 19 5000 A2DP
SET CONTROL BAUD 115200,8n1
SET CONTROL BIND 0 80 R CALL %0 17 AVRCP
SET CONTROL CD 00 0
SET CONTROL CODEC SBC JOINT_STEREO 44100 0
SET CONTROL ECHO 7
SET CONTROL ESCAPE 43 00 1
SET CONTROL GAIN 8 8
SET CONTROL MICBIAS b 0
SET CONTROL MSC DTE 00 00 00 00 00 00
SET CONTROL PIO 00 00
SET CONTROL PREAMP 1 1
SET CONTROL READY 00
SET CONTROL RINGTONE 6,gfgfgf__gfgfgf______gfgfgf__gfgfgf
SET CONTROL VREGEN 1 00
SET PROFILE A2DP SINK
SET PROFILE HFP Volvo HF
SET PROFILE SPP mu uart
SET PROFILE PBAP ON
SET PROFILE AVRCP CONTROLLER
SET
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
 **59546> Accept port 23, conn=0x3fff6db8, pool slot 0**
SYNTAX ERROR
at

**59546> Accept port 23, conn=0x3fff6db8, pool slot 0** - info about next connection..
sys log setup to Filter EMERG.

Error apper when 3'th client connect to AP..
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
205483> HTTP GET /services/info: 200, 54ms, h=19576
SYNTAX ERROR
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
241776> Accept port 23, conn=0x3fff6db0, pool slot 0
SYNTAX ERROR
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
[... 216 lines cut ...]
SYNTAX ERROR
SYNTAX ERROR
SYNTAX ERROR
SYNTAX ERROR
549934> E:M 2928
SYNTAX ERROR
549949> espbuffsend: cannot alloc tx buffer
SYNTAX ERROR
549961> E:M 2928
SYNTAX ERROR
WRAP THOR AI (5.0.2 build 992)
Copyright (c) 2003-2013 Bluegiga Technologies Inc.
READY.
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED
NO CARRIER 0 ERROR 9801 L2CAP_CONNECTION_FAILED

Garbage reset WT32 module.,.

Soft-AP Max Connections = 4.

How much user can connect to the bridge?
I have to connect 2 users or more (3?) and send data bidirectional (tcp-uart, uart-tcp)..
How long socket is keeping alive? When server disconnect the client?
I got remote controller sending some data to wt32, and server disconnect remote controller when some time left..
Can I keep this connection?

wchpikus, 2017-06-03:
ok i thing i got the problem with tx buffer...

[IMG]http://i65.tinypic.com/2d7tyxs.png[/IMG]

I have to disable debug log..323842> Turning OFF uart log

Now no errors..
I still waiting for keep connection...I heard, ESP SDK has 5min unused socket close to save memory..
Is it true?

My client, when socket closed from server side , like to send some data, returned fatal error, "socket not connect" and reboot:)
Any idea to fix or any solution for this situation?

wchpikus, 2017-06-04:
hmm, no, i can not fix it..
Noiw i check again, on debug lo (www side on esp) i see:
275126> HTTP GET /log/text: 200, 54ms, h=19208
285403> Accept port 23, conn=0x3fff7480, pool slot 2
285421> E:M 2928
285421> espbuffsend: cannot alloc tx buffer
285421> E:M 2928
285421> espbuffsend: cannot alloc tx buffer
this is second connection of ip client..

wchpikus, 2017-06-04:
it is happens when client disconnect and connect again...
Maybe some logs helps...

0 excvaddr=0x0 depc=0x0
   120> Flash map 4MB:512/512, manuf 0xE0 chip 0x4016
   120> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62: ready, heap=25112
   121> initializing user application
   121> Waiting for work to do...
   121> mode : softAP(a2:20:a6:00:7c:8e)
   121> add if1
   122> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   124> bcn 100
 15119> Wifi check: mode=AP status=255
 24621> add 1
 24621> aid 1
 24621> station: 20:10:09:28:00:08 join, AID = 1
 24623> Wifi AP: station 20:10:09:28:00:08 joined, AID = 1
 28172> HTTP GET /pins: 200, 76ms, h=17032
 28222> HTTP GET /wifi/info: 200, 85ms, h=19056
 28237> HTTP GET /system/info: 200, 70ms, h=21080
 34262> HTTP GET /log/text: 200, 14ms, h=18152
 34301> HTTP GET /log/dbg: 200, 56ms, h=20192
 39240> HTTP GET /system/info: 200, 4987ms, h=20720
 60605> add 2
 60606> aid 2
 60606> station: 5c:cf:7f:f7:56:b6 join, AID = 2
 60607> Wifi AP: station 5c:cf:7f:f7:56:b6 joined, AID = 2
 61446> Accept port 23, conn=0x3fff7448, pool slot 0
 61447> Accept port 23, conn=0x3fff72a0, pool slot 1
 84089> HTTP GET /log.html: 200, 108ms, h=11000
 84173> HTTP GET /console.js: 200, 107ms, h=13064
 84273> HTTP GET /style.css: 200, 214ms, h=15208
 84276> HTTP GET /ui.js: 200, 213ms, h=16984
 84322> HTTP GET /pure.css: 200, 267ms, h=19136
 84449> HTTP GET /log/dbg: 200, 79ms, h=17200

277905> HTTP GET /favicon.ico: 200, 153ms, h=16624
305780> HTTP GET /log.html: 200, 111ms, h=8336
305883> HTTP GET /console.js: 200, 115ms, h=10392
305987> HTTP GET /style.css: 200, 229ms, h=12536
305992> HTTP GET /ui.js: 200, 228ms, h=14408
306033> HTTP GET /pure.css: 200, 285ms, h=16616
306142> HTTP GET /log/dbg: 200, 55ms, h=14576
306186> HTTP GET /log/text: 200, 7ms, h=14576
306189> HTTP GET /favicon.ico: 200, 111ms, h=16272
314159> station: 5c:cf:7f:f7:56:b6 leave, AID = 2
314159> rm 2
314160> Wifi AP: station 5c:cf:7f:f7:56:b6 left, AID = 2
314160> add 2
314160> aid 2
314161> station: 5c:cf:7f:f7:56:b6 join, AID = 2
314162> Wifi AP: station 5c:cf:7f:f7:56:b6 joined, AID = 2
314997> Accept port 23, conn=0x3fff72c8, pool slot 0
314999> Accept port 23, conn=0x3fff7338, pool slot 1
330238> station: 5c:cf:7f:f7:56:b6 leave, AID = 2
330239> rm 2
330239> Wifi AP: station 5c:cf:7f:f7:56:b6 left, AID = 2
330240> add 2
330240> aid 2
330240> station: 5c:cf:7f:f7:56:b6 join, AID = 2
330241> Wifi AP: station 5c:cf:7f:f7:56:b6 joined, AID = 2
331029> HTTP GET /log.html: 200, 108ms, h=7568
331116> HTTP GET /console.js: 200, 107ms, h=9528
331231> HTTP GET /style.css: 200, 231ms, h=11768
331234> HTTP GET /ui.js: 200, 229ms, h=13624
331269> HTTP GET /pure.css: 200, 273ms, h=15848
331369> HTTP GET /log/dbg: 200, 54ms, h=13816

c tx buffer
370869> E:M 2928
370869> espbuffsend: cannot alloc tx buffer
370869> E:M 2928
370870> espbuffsend: cannot alloc tx buffer
370870> E:M 2928
370870> espbuffsend: cannot alloc tx buffer
370870> E:M 2928
370870> espbuffsend: cannot alloc tx buffer
370870> E:M 2928
370870> espbuffsend: cannot alloc tx buffer
370870> E:M 2928
370870> espbuffsend: cannot alloc tx buffer
370882> E:M 2928
370882> espbuffsend: cannot alloc tx buffer
370882> E:M 2928
370882> espbuffsend: cannot alloc tx buffer
370883> E:M 2928
370883> espbuffsend: cannot alloc tx buffer
370883> E:M 2928
370883> espbuffsend: cannot alloc tx buffer
370883> E:M 2928
370883> espbuffsend: cannot alloc tx buffer
370883> E:M 2928
370883> espbuffsend: cannot alloc tx buffer
379336> E:M 2928
379336> espbuffsend: cannot alloc tx buffer
379339> E:M 2928
379339> espbuffsend: cannot alloc tx buffer
379503> E:M 2928
379503> espbuffsend: cannot alloc tx buffer
379509> E:M 2928
379509> espbuffsend: cannot alloc tx buffer
379512> E:M 2928
379512> espbuffsend: cannot alloc tx buffer
379516> E:M 2928
379516> espbuffsend: cannot alloc tx buffer
379520> E:M 2928
379520> espbuffsend: cannot alloc tx buffer
379523> E:M 2928
379523> espbuffsend: cannot alloc tx buffer
379599> E:M 2928
379599> espbuffsend: cannot alloc tx buffer
379688> E:M 2928
379688> espbuffsend: cannot alloc tx buffer

tve, 2017-06-05:
`E:M 2928` means that malloc tried to allocate 2928 bytes and couldn't. That's the size of a buffer used by TCP. I'm afraid the data rate and number of connections simply exceeds the amount of buffer space available in esp-link :-(. One thing you could try is to switch to liblwip_536, which uses smaller TCP packets and therefore requires less buffer space. I'm planning to compile the next release with that, if it's available (Espressif hasn't always made that lib available).

tve, 2017-06-05:
Do you get the problem looking at the esp-link log, at the web console page for the uC, or only having TCP connections running?

wchpikus, 2017-06-05:
About connection client-server I do not have now any problems, only with this, that i have to sent "ping" one for 5 min to keepalive connection.
Www logs on serwer works ok, on another client i got some lag or sometimes no data is recived, i will try to debug this.

I ask about closing socket, but i found solution.;

"One thing you could try is to switch to liblwip_536, which uses smaller TCP packets and therefore requires less buffer space."

I do not know what you write to me;)
This is sdk lib or something?;)

wchpikus, 2017-06-06:
I tested again, error 370870> espbuffsend: cannot alloc tx buffer still appear, when second client connect to AP.
Scenario:
client 1 connected to AP, communication uart-wifi work ok.
client 2 connecting to AP, communication client 1 to AP disappear, error TX buffor on log.
Communication client 2 uart-wifi works ok .

I like to connect more than one client and listen uart response...

tve, 2017-06-06:
> About connection client-server I do not have now any problems, only with this, that i have to sent "ping" one for 5 min to keepalive connection.

Yup, that's built into the Espressif SDK. It can be changed in code, but esp-link doesn't provide a UI for that.

> espbuffsend: cannot alloc tx buffer

it runs out of memory, the esp8266 doesn't have a lot... IIRC the TCP window size used by the SDK is 5680 bytes, so if ACKs are coming back fast enough that's how much memory each connection takes, not counting an extra buffer for the next packet.

wchpikus, 2017-06-06:
I tryed another brigde, and no problem...but without www side, only AT commands.
2 client connected no problem, this is only sent the same bufor to another connection..
Mayby i wrong..using one bufer and cast to clients.

wchpikus, 2017-06-08:
Could you check, is it possible to make one bufer to tx from uart and send them to all clients?
It could be option enable on www side.
So if you like to send tx to one of them ( who send request to uart e.g AT) server send  response from tx to sender or all connected clients.
This option reduce create  bufors for e
ach clients.

Now if i understend each client has one bufor for tx data yes?

tve, 2017-06-16:
Each TCP connection fundamentally has its own retransmit buffers. There is no way around that...

tve, 2017-06-18:
You could try the latest 3.2 alpha, it uses smaller TCP buffer sizes, that may do the trick..